moveit/moveit #2130: catkin_make breaks when trying to install moveit from source

evangelosc, 2020-06-02:
### Description

I get a compilation error when trying to compile moveit from source.

### Your environment
* ROS Distro: [Melodic]
* OS Version: Ubuntu 18.04
* Source or Binary build? Source
* If source, which branch? melodic-devel

### Steps to reproduce
- rosdep update  
- sudo apt-get update
- sudo apt-get dist-upgrade 
- sudo apt-get install python-wstool python-catkin-tools clang-format-3.9 
- mkdir ~/ws_moveit
- cd ~/ws_moveit 
- source /opt/ros/melodic/setup.bash
- wstool init src
- wstool merge -t src https://raw.githubusercontent.com/ros-planning/moveit/master/moveit.rosinstall
- wstool update -t src
- rosdep install -y --from-paths src --ignore-src --rosdistro ${ROS_DISTRO}
- catkin config --extend /opt/ros/${ROS_DISTRO} --cmake-args -DCMAKE_BUILD_TYPE=Release
- catkin_make

### Expected behaviour
Install MoveIT from source 

### Actual behaviour
catkin_make breaks and i get an error that a few scripts under moveit/moveit_ros/move_group do not compile appropriately

### Backtrace or Console output
[ 80%] Built target moveit_robot_interaction
[ 80%] Built target benchmark_ik
Scanning dependencies of target move_group
[ 80%] Linking CXX executable /home/evangelos/ws_moveit/devel/lib/moveit_ros_warehouse/moveit_warehouse_broadcast
[ 80%] Building CXX object moveit/moveit_ros/warehouse/warehouse/CMakeFiles/moveit_init_demo_warehouse.dir/src/initialize_demo_db.cpp.o
[ 80%] Linking CXX executable /home/evangelos/ws_moveit/devel/lib/moveit_ros_warehouse/moveit_warehouse_services
Scanning dependencies of target moveit_move_group_default_capabilities
[ 80%] Building CXX object moveit/moveit_ros/warehouse/warehouse/CMakeFiles/moveit_save_to_warehouse.dir/src/save_to_warehouse.cpp.o
[ 80%] Building CXX object moveit/moveit_ros/warehouse/warehouse/CMakeFiles/moveit_warehouse_save_as_text.dir/src/save_as_text.cpp.o
[ 80%] Building CXX object moveit/moveit_ros/warehouse/warehouse/CMakeFiles/moveit_warehouse_import_from_text.dir/src/import_from_text.cpp.o
[ 80%] Linking CXX executable /home/evangelos/ws_moveit/devel/lib/moveit_ros_warehouse/moveit_init_demo_warehouse
[ 80%] Building CXX object moveit/moveit_ros/benchmarks/CMakeFiles/moveit_ros_benchmarks.dir/src/BenchmarkOptions.cpp.o
[ 80%] Linking CXX executable /home/evangelos/ws_moveit/devel/lib/moveit_ros_warehouse/moveit_save_to_warehouse
[ 80%] Building CXX object moveit/moveit_ros/move_group/CMakeFiles/move_group.dir/src/move_group.cpp.o
[ 80%] Linking CXX executable /home/evangelos/ws_moveit/devel/lib/moveit_ros_warehouse/moveit_warehouse_import_from_text
[ 81%] Linking CXX executable /home/evangelos/ws_moveit/devel/lib/moveit_ros_warehouse/moveit_warehouse_save_as_text
[ 81%] Building CXX object moveit/moveit_ros/benchmarks/CMakeFiles/moveit_ros_benchmarks.dir/src/BenchmarkExecutor.cpp.o
[ 81%] Linking CXX executable /home/evangelos/ws_moveit/devel/lib/moveit_ros_move_group/move_group
[ 82%] Linking CXX shared library /home/evangelos/ws_moveit/devel/lib/libmoveit_ros_benchmarks.so
[ 82%] Built target test_kinematics_plugin
Scanning dependencies of target moveit_pick_place_planner
[ 82%] Building CXX object moveit/moveit_ros/move_group/CMakeFiles/moveit_move_group_default_capabilities.dir/src/default_capabilities/move_action_capability.cpp.o
[ 82%] Building CXX object moveit/moveit_ros/manipulation/pick_place/CMakeFiles/moveit_pick_place_planner.dir/src/pick_place_params.cpp.o
[ 84%] Building CXX object moveit/moveit_ros/manipulation/pick_place/CMakeFiles/moveit_pick_place_planner.dir/src/manipulation_pipeline.cpp.o
[ 84%] Built target moveit_warehouse_services
[ 84%] Building CXX object moveit/moveit_ros/move_group/CMakeFiles/moveit_move_group_default_capabilities.dir/src/default_capabilities/plan_service_capability.cpp.o
[ 84%] Building CXX object moveit/moveit_ros/manipulation/pick_place/CMakeFiles/moveit_pick_place_planner.dir/src/reachable_valid_pose_filter.cpp.o
[ 84%] Built target moveit_warehouse_broadcast
[ 84%] Building CXX object moveit/moveit_ros/manipulation/pick_place/CMakeFiles/moveit_pick_place_planner.dir/src/approach_and_translate_stage.cpp.o
[ 84%] Building CXX object moveit/moveit_ros/manipulation/pick_place/CMakeFiles/moveit_pick_place_planner.dir/src/plan_stage.cpp.o
[ 84%] Built target moveit_ros_benchmarks
Scanning dependencies of target moveit_run_benchmark
[ 84%] Built target moveit_init_demo_warehouse
[ 84%] Building CXX object moveit/moveit_ros/move_group/CMakeFiles/moveit_move_group_default_capabilities.dir/src/default_capabilities/execute_trajectory_action_capability.cpp.o
[ 84%] Building CXX object moveit/moveit_ros/benchmarks/CMakeFiles/moveit_run_benchmark.dir/src/RunBenchmark.cpp.o
[ 84%] Built target moveit_warehouse_import_from_text
[ 84%] Built target moveit_visual_tools
[ 84%] Built target moveit_warehouse_save_as_text
[... 42 lines cut ...]
moveit/moveit_ros/move_group/CMakeFiles/moveit_move_group_default_capabilities.dir/build.make:182: recipe for target 'moveit/moveit_ros/move_group/CMakeFiles/moveit_move_group_default_capabilities.dir/src/default_capabilities/state_validation_service_capability.cpp.o' failed
make[2]: *** [moveit/moveit_ros/move_group/CMakeFiles/moveit_move_group_default_capabilities.dir/src/default_capabilities/state_validation_service_capability.cpp.o] Error 4
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
moveit/moveit_ros/manipulation/pick_place/CMakeFiles/moveit_pick_place_planner.dir/build.make:206: recipe for target 'moveit/moveit_ros/manipulation/pick_place/CMakeFiles/moveit_pick_place_planner.dir/src/pick.cpp.o' failed
make[2]: *** [moveit/moveit_ros/manipulation/pick_place/CMakeFiles/moveit_pick_place_planner.dir/src/pick.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
moveit/moveit_ros/manipulation/pick_place/CMakeFiles/moveit_pick_place_planner.dir/build.make:134: recipe for target 'moveit/moveit_ros/manipulation/pick_place/CMakeFiles/moveit_pick_place_planner.dir/src/approach_and_translate_stage.cpp.o' failed
make[2]: *** [moveit/moveit_ros/manipulation/pick_place/CMakeFiles/moveit_pick_place_planner.dir/src/approach_and_translate_stage.cpp.o] Error 4
CMakeFiles/Makefile2:21039: recipe for target 'moveit/moveit_ros/manipulation/pick_place/CMakeFiles/moveit_pick_place_planner.dir/all' failed
make[1]: *** [moveit/moveit_ros/manipulation/pick_place/CMakeFiles/moveit_pick_place_planner.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:20891: recipe for target 'moveit/moveit_ros/move_group/CMakeFiles/moveit_move_group_default_capabilities.dir/all' failed
make[1]: *** [moveit/moveit_ros/move_group/CMakeFiles/moveit_move_group_default_capabilities.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

gavanderhoorn, 2020-06-02:
It's very likely your computer is running out of memory while compiling MoveIt.

Try reducing the parallelism by running `catkin_make -jN` where `N` is some number `<` the nr of cores you have in your machine.

So if you have 4 cores, try `catkin_make -j2`.

Note that this will result in longer compilation times.

evangelosc, 2020-06-02:
Thanks @gavanderhoorn 
It was running out of memory. catkin_make didn't break when I went for the lower cores thing.

closed 2020-06-02 by evangelosc