moveit/moveit #2143: IKfast not working

hs2041, 2020-06-06:
Hi, 
I was trying to create my own ikfast plugin for a model based on the UR5. I am using the following page as reference: http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/ikfast/ikfast_tutorial.html
My urdf file is:

<?xml version="1.0" ?>

<robot name="add_cobot" xmlns:xacro="http://ros.org/wiki/xacro">

  <link name="base_link"/>

  <link name="link_1"/>

  <link name="link_2"/>

  <link name="link_3"/>

  <link name="link_4"/>

  
  <link name="link_5"/>

  <link name="link_6"/>

  <joint name="joint_1" type="revolute">
    <origin rpy="0 0 0" xyz="0 0 0.376"/>
    <axis xyz="0 0 1"/>
    <parent link="base_link"/>
    <child link="link_1"/>
    <limit effort="0" lower="-3.14" upper="3.14" velocity="1.7453"/>
  </joint>
  <joint name="joint_2" type="revolute">
    <origin rpy="0 0 0" xyz="0 0 0"/>
    <axis xyz="1 0 0"/>
    <parent link="link_1"/>
    <child link="link_2"/>
        <limit effort="0" lower="-3.14" upper="3.14" velocity="1.7453"/>
  </joint>
  <joint name="joint_3" type="revolute">
    <origin rpy="0 0 0" xyz="0.0 0.231 0.0"/>
    <axis xyz="1 0 0"/>
    <parent link="link_2"/>
    <child link="link_3"/>
        <limit effort="0" lower="-3.14" upper="3.14" velocity="1.7453"/>
  </joint>
  <joint name="joint_4" type="revolute">
    <origin rpy="0 0 0" xyz="-0.115 0.264 0.0"/>
    <axis xyz="1 0 0"/>
    <parent link="link_3"/>
    <child link="link_4"/>
        <limit effort="0" lower="-3.14" upper="3.14" velocity="1.7453"/>
  </joint>
  <joint name="joint_5" type="revolute">
    <origin rpy="0 0 0" xyz="0.0 0 0.0926"/>
    <axis xyz="0 0 1"/>
    <parent link="link_4"/>
    <child link="link_5"/>
        <limit effort="0" lower="-3.14" upper="3.14" velocity="1.7453"/>
  </joint>
  <joint name="joint_6" type="revolute">
    <origin rpy="0 0 0" xyz="0.0926 0 0.0"/>
    <axis xyz="1 0 0"/>
    <parent link="link_5"/>
    <child link="link_6"/>
        <limit effort="0" lower="-3.14" upper="3.14" velocity="1.7453"/>
  </joint>
</robot>

I am able to generate the ikfast plugin, however, the inverse kinematics part doesn't behave correctly, joint angle 5 is always 0. I am attaching a screenshot of the terminal for your reference.
[image]

As you can see in the ik solution 6, all the joint angles are correct except joint 5. Please help!

felixvd, 2020-07-17:
I haven't used IKFast, but `joint_2` in your URDF is at `<origin rpy="0 0 0" xyz="0 0 0"/>`. If that doesn't help, try with a URDF that you know works, like [the panda](https://github.com/frankaemika/franka_ros/blob/kinetic-devel/franka_description/robots/panda_arm.xacro).

hs2041, 2020-07-23:
Hi Felix, 
Thanks for input but the joint_2 has its axis along the x-axis. I have checked the urdf in rviz and openrave it's working fine. Only the IKfast plugin is giving me a headache