moveit/moveit #2145: Attach the object to a fixed coordinate system in the Moveit Scene, but they move.

seoulstudent, 2020-06-09:
### Description

Hello,

I'm using a mobile robot with a manipulator attached to it. (ex, Husky with UR5)

I made a Moveit Config file and I'm adding objects to the Planning Scene(Rviz) by using function (e.g add_mesh, add_box....)

But even if I put the 'a fixed frame'(e.g map) in ' box/mesh header.frame_id', box/mesh in rviz move with (follows) mobile robot which I gave the speed command using ros_control. I made sure 'map frame' didn't move.

My robot coordinate system is 'map ~ odom ~ baselink ~  '.

Is it possible to hold the object in Planning Scene when I use mobile+manipulator?

I’ll be waiting for your reply.

Thank you

### Your environment
* ROS Distro: [Kinetic]
* OS Version: e.g. Ubuntu 16.04
* Source or Binary build? Source
* If binary, which release version?
* If source, which branch? Kinetic

### Steps to reproduce
This is a problem with mobile robot + manipulator. When I use add_box or add_mesh, even if I give  header.frame_id a fixed frame (ex. box_pose.header.frame_id = "map"), box or mesh follows mobile robot which I gave speed command

v4hn, 2020-06-09:
There is no need to attach anything in your scenario. Any object you add to the planning scene will be "attached" to the configured planning frame.

If you want the objects to be static in the map frame, you have to make sure the planning frame is actually `map`. This should be the case if you configured a virtual joint from `map` to `base_link`.

closed 2020-06-09 by v4hn

seoulstudent, 2020-06-15:
Dear sir, 

Sorry for the delay in getting back to you. 

Thank you for your prompt reply.

I want to express my appreciation for your valuable advice.

I hope everything is fine with you.