moveit/moveit #2149: OMPL Collision Checking for Trajectories Planned in Cartesian State Space

marip8, 2020-06-15:
**Is your feature request related to a problem? Please describe.**

I'm working on an application where I need to plan constrained free-space motions to navigate a large end-effector into a tight space. I have been experimenting with planning in OMPL using Cartesian orientation path constraints (glass-upright, specifically) and Cartesian goal constraints. From my understanding, when Cartesian path constraints are present, MoveIt chooses to represent the OMPL planning problem in Cartesian space (rather than joint space). This works fairly well for my planning problem; however, wrist flips occasionally occur between adjacent planned waypoints, producing collisions that are not detected. It seems like when the OMPL state space is represented by Cartesian space, trajectory interpolation for collision checking (using the `longest_valid_segment_fraction` and `maximum_waypoint_distance` parameters) also occurs in Cartesian space. Thus, collisions caused by joint flips (i.e. IK solutions with different discrete configurations at adjacent interpolated waypoints) are not detected.

Video to come, pending approval from client

**Describe the solution you'd like**

I would expect collision detection for OMPL plans to occur in joint space regardless of the representation of the planning space because robots physically move in joint space

**Describe alternatives you've considered**

I'm not sure there is an alternative besides changing the behavior of the collision detection described above, but I have tried representing my problem in a couple different ways to achieve the desired results while still checking collisions in joint space

1. Enforce joint state space representation
    - Planning for my problem succeeds with much lower probability when planning in joint space due to the path constraint
1. Enforce joint state space representation and use an approximate constraint manifold database
   - Fairly reliable, but still less so than planning in Cartesian space (no metrics here)
   - The path itself is still planned in joint space, so it retains a lot of the strange un-intuitive motions characteristic of joint-space sampling based planners. Planning in Cartesian space seems to produce much more intuitive looking motions, which is important to the stakeholders of my project
   - The executable for making the approximate constraint manifold database also doesn't seem to exist anymore in the master branch (possibly deprecated in favor of OMPL constrained planning)

**Additional context**
This seems tangentially related to #1969 and #2092. If there is work happening to change/improve the behavior of OMPL for trajectories planned in Cartesian space, this might issue might be something to consider

JeroenDM, 2020-06-16:
> It seems like when the OMPL state space is represented by Cartesian space, trajectory interpolation for collision checking (using the longest_valid_segment_fraction and maximum_waypoint_distance parameters) also occurs in Cartesian space. 

Interesting analysis, I had not come across this aspect of the problem.

In general, it's a long-standing issue that I certainly will try to solve this summer. However, it might require changes beyond the current OMPL interface, as discussed in [this talk](https://www.youtube.com/watch?v=aM5Ai_O12KY&list=PL2V08eX0CdgCazaFSjXGhfn3vcFAUH8NS&index=6) on the IK interface by Robert Haschke.

Other related links:
- #562 
- #541
- [ROS Answers question with video](https://answers.ros.org/question/203662/planning-with-orientation-constraint-in-moveit/)

v4hn, 2020-06-16:
Yes, this is a long-standing issue in the `PoseModelStateSpace` which is not entirely easy to fix.

Our lab's local workaround is usually #541 and [moveit_generate_state_database](https://ros-planning.github.io/moveit_tutorials/doc/planning_with_approximated_constraint_manifolds/planning_with_approximated_constraint_manifolds_tutorial.html).

The code somehow assumes that IK solutions found for neighboring Poses in workspace are always close, which is simply not true unless you configured your special-purpose IK solver to rely on gradients only and not do restarts. @JeroenDM is right in that a changed IK interface would improve
the general situation here.

The alternative fix, to interpolate IK solutions of neighboring states and require a maximum deviation from the straight-line path, would ensure that you don't see any broken paths anymore.
You would still need to check that you actually still receive *correct* solutions because the same state does not represent all different IK solutions - that would add another dimension of complexity.
A patch for this fix would still be very welcome!

This bug is one of the reasons why I'm looking forward to the outcome of @JeroenDM's GSoC project documented in #2092 .

JeroenDM, 2020-08-28:
While we hopefully have [an extra alternative](https://github.com/ros-planning/moveit/pull/22730) in the near future, I still feel bad for not patching `PoseModelStateSpace` :( If anyone else wants to tackle it, feel free to reach out to me for help. (I feel like I can write a book about the current ompl interface, but I tactically avoided the `PoseModelStateSpace` part...)

> You would still need to check that you actually still receive correct solutions because the same state does not represent all different IK solutions - that would add another dimension of complexity.

I do not entirely understand this sentence. A state in `PoseModelStateSpace` associates a specific pose with a single joint solution, at least I think. But what do you mean by "correct solutions" in this context, and who needs to check this?