moveit/moveit #2153: moveit_setup_assistant support for xacro files

thejose5, 2020-06-17:
**Is your feature request related to a problem? Please describe.**
Since the current moveit_setup_assistant only supports urdf files, xacro files have to be first converted to urdf before they can be used with moveit. Since xacro is a common robot model format and the easiest to use, it would be great if this conversion could happen internally within moveit_setup_assistant.

**Describe the solution you'd like**
I would like the option to upload a xacro file instead of a urdf file in the first page of the moveit_setup_assistant gui after selecting "create new moveit configuration package".

**Additional Information**
I am opening this issue since I would like to solve it myself. This is the first time I will be contributing towards a ROS feature. Therefore, any help and input is welcome.

gavanderhoorn, 2020-06-18:
> Since the current moveit_setup_assistant only supports urdf files, xacro files have to be first converted to urdf before they can be used with moveit.

actually, the MSA supports loading `.xacro` files directly.

Perhaps you've run into a problem with this which made you conclude `.xacro` is not supported.

Could you please clarify?

thejose5, 2020-06-18:
My apologies. There was a problem with my xacro file and I assumed that xacro files weren't supported.

closed 2020-06-18 by thejose5