moveit/moveit #2156: Python3 roscpp_init: dynamic module does not define module export function

a-z-e-r-i-l-a, 2020-06-19:
**Is your feature request related to a problem? Please describe.**
I would like to invoke roscpp_init('node_name', []) that is mentioned [here](http://wiki.ros.org/ROS/Tutorials/Using%20a%20C%2B%2B%20class%20in%20Python) in Python3.

If I do so however I get the following error:

```
    from moveit_ros_planning_interface._moveit_roscpp_initializer import roscpp_init
ImportError: dynamic module does not define module export function (PyInit__moveit_roscpp_initializer)
```
I thought if I build moveit from source in my workspace with cmake arguments corresponding python3 it may fix the problem, but even without these arguments haven't been able to build the planning_interface package without error. 

I would be glad if you could kindly help me about this issue.

IndustrialEngStudent, 2020-08-19:
I have the same error when trying to import moveit_commander in the python based move group interface. Did you solve your issue?

a-z-e-r-i-l-a, 2020-08-19:
@IndustrialEngStudent not through moveit. but in the end I made a package that would initialize the node handle similar to how it's done in `moveit_ros_planning_interface._moveit_roscpp_initializer import roscpp_init `, and then built this package with python3 cmake args.

rhaschke, 2020-08-19:
Building and using MoveIt's python API with Python3 works as far as I know. Of course, you need to build your _whole_ ROS environment with Python3 support - as done in Noetic.

closed 2020-08-28 by rhaschke