moveit/moveit #2161: Allow MoveIt to 'avoid' planning when start->end is collision free

gavanderhoorn, 2020-06-23:
This was brought up by @pbeeson in [Can MoveIt first check shortest path before planning?](https://answers.ros.org/question/355617) on ROS Answers.

As he does a good job of describing what this should do, a quote (I paraphrased slightly to summarise the discussion under the answer there):

> Is there a way to ask MoveIt! to check the simple 2 point start/goal trajectory for collision *first*, before calling OMPL planners that might take seconds to find a PRM based solution only to then shorten it back to 2 points?

This seems like a useful shortcut.

@pbeeson then suggested to do this in a Planning Request Adapter, which could be configured as part of a planning pipeline and would filter (almost) incoming planning requests according to the description above (so: collision free -> linearly interpolate, return).

Whether this would be possible as a Planning Adapter would be something to check.

Perhaps the multiple planner support (#2127) could support this use-case by asking a planner to do this (such as the Pilz planners) instead of having an adapter 'play planner'.

rhaschke, 2020-06-23:
I like this idea very much. It should be possible to implement this as a planning adapter. However, as we cannot avoid calling the remaining adapters (and the actual planner), my suggestion is to modify the planning request for the remaining pipeline to plan from goal -> goal only. The OMPL planners should be clever enough to notice that there is no planning needed. @mamoll can you confirm this?

pbeeson, 2020-06-23:
@rhaschke Would this then also bypass any STOMP post-processing that you might still want on the plan?

rhaschke, 2020-06-23:
This depends on the planner. I'm not sure about STOMP, but CHOMP can be used as a post-processing planning adapter. This would be executed.

pbeeson, 2020-06-23:
I guess what I was wondering specifically was if you changed the plan to goal->goal, would this affect post-processing adapters?

rhaschke, 2020-06-23:
I would say that depends on the order of the adapters only.

v4hn, 2020-06-23:
> It should be possible to implement this as a planning adapter. However, as we cannot avoid calling the remaining adapters (and the actual planner)

Of course you can, just don't call further into the pipeline. It means the plugin needs to be the last one before the planner, but that can be documented.

> I like this idea very much.

I do not think I do.

If you ask MoveIt to plan with a specific planner, I expect that planner to be called.

For longer distances, the shortest path in joint-space can also be quite an ugly path for the end-effector in Cartesian space, even if it is collision-free.
Especially optimizing planners can have different costs than c-space distance.

Also, this is a standard use-case for MTC's [Fallback container](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/include/moveit/task_constructor/container.h#L170) when using a [JointInterpolationPlanner](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/include/moveit/task_constructor/solvers/joint_interpolation.h) as first option.

> my suggestion is to modify the planning request for the remaining pipeline to plan from goal -> goal only.

That's an unnecessary ugly workaround and you would still incur setup overhead, even if the planner does not run.

rhaschke, 2020-06-24:
> > We cannot avoid calling the remaining adapters (and the actual planner).

> Of course you can, just don't call further into the pipeline. It means the plugin needs to be the last one before the planner, but that can be documented.

This was my first reflex as well. But that means that the timing adapter will not be called. IMHO this latter one should be the last being called in any case.

v4hn, 2020-06-24:
> But that means that the timing adapter will not be called.

Nonsense. It just depends on the order of the adapters.

> IMHO this latter one should be the last being called in any case.

As "first" and "last" are confusing in this context (by programm flow the first one is also the last one), let's use "outermost" and "innermost" instead.

The timing adapter (whichever is used) should almost always be the *outermost* one to account for modifications by other plugins.
The potential adapter discussed here should definitely be the *innermost* one, as it would act like an actual planner.

But again, while you can obviously write such an adapter, I explicitly veto adding it to the defaults.

gavanderhoorn, 2020-06-24:
> But again, while you can obviously write such an adapter, I explicitly veto adding it to the defaults.

The linked ROS Answers Q&A and the discussion here do not suggest anything like that Michael.

We're simply discussing the technical side here.

Whether this is something that should then be used would be up to the user, which is exactly why we'd implement it in an adapter.

v4hn, 2020-06-24:
> which is exactly why we'd implement it in an adapter.

Go ahead. It's a nice piece of code to have and it's straight-forward to implement.

JeroenDM, 2020-07-15:
I don't know the details of planning adapters, but if someone implements this, the code will probably be similar to the [lerp_motion_planner](https://github.com/ros-planning/moveit_tutorials/tree/2ebb32edd2160dd9975f00c410b2a8b2542e0ec4/doc/creating_moveit_plugins/lerp_motion_planner) from the tutorial "[Creating MoveIt Plugins](https://ros-planning.github.io/moveit_tutorials/doc/creating_moveit_plugins/plugin_tutorial.html)".

felixvd, 2021-02-19:
This probably isn't the best issue for newcomers, but someone with experience could probably get something working within a day.