moveit/moveit #2164: building melodic branch with python3 cmake arguments

a-z-e-r-i-l-a, 2020-06-25:
I have an aplication which uses moveit_ros_planning_interface but when I use it in python3 the following error is given:

```
    from .roscpp_initializer import *
  File "/opt/ros/melodic/lib/python2.7/dist-packages/moveit_commander/roscpp_initializer.py", line 35, in <module>
    from moveit_ros_planning_interface import _moveit_roscpp_initializer
ImportError: dynamic module does not define module export function (PyInit__moveit_roscpp_initializer)
```
* ROS Distro: [Melodic]
* OS Version: Ubuntu 18.04
* Source
* melodic-devel branch

For this I cloned moveit and moveit_resources in my workspace and proceeded with catkin build using the following cmake arguments that are for python3:

`catkin config --cmake-args   -DPYTHON_EXECUTABLE=/usr/bin/python3   -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m   -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so
`
the build finishes successfully however when using the application now the following error comes:` Segmentation fault (core dumped)`.

Do you know a work around this issue that I can apply? in addition, I build the melodic-devel branch of moveit.

YeeCY, 2020-07-08:
Same problem when I use python3.7 in an anaconda environment.

`ImportError: dynamic module does not define module export function (PyInit__moveit_roscpp_initializer)`

rhaschke, 2020-07-08:
The error `PyInit__moveit_roscpp_initializer` is related to python2 extension lib to be loaded in python3. This will not work.
Hence, you need to ensure - as @a-z-e-r-i-l-a did - that the libs are compiled against python3.

As [this Noetic CI build](https://travis-ci.com/github/ros-planning/moveit/jobs/357433316) illustrates, the master branch successfully builds with a python3 environment. So, please check the logs, to know which python version is actually found by cmake.

Note: ROS1, up to Melodic, is not intended to be used via python3. Only Noetic provides python3 support.

YeeCY, 2020-07-08:
Actually, I tried to build the moveit from scratch like tf2_py in the following links for Python 3, but I still got the same error above.

https://github.com/ros/geometry2/issues/259
https://answers.ros.org/question/326226/importerror-dynamic-module-does-not-define-module-export-function-pyinit__tf2/

YeeCY, 2020-07-08:
And here is part of my catkin_make log, I think python3 is recognized when building

```
Base path: /home/yee/mujoco_ros
Source space: /home/yee/mujoco_ros/src
Build space: /home/yee/mujoco_ros/build
Devel space: /home/yee/mujoco_ros/devel
Install space: /home/yee/mujoco_ros/install
####
#### Running command: "make cmake_check_build_system" in "/home/yee/mujoco_ros/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/yee/mujoco_ros/devel
-- Using CMAKE_PREFIX_PATH: /home/yee/mujoco_ros/devel;/home/yee/bulldog_ws/devel;/home/yee/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/yee/mujoco_ros/devel;/home/yee/bulldog_ws/devel;/home/yee/catkin_ws/devel;/opt/ros/kinetic
-- Found PythonInterp: /home/yee/anaconda3/envs/robot/bin/python (found suitable version "3.7.7", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /home/yee/anaconda3/envs/robot/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/yee/mujoco_ros/build/test_results
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found PythonInterp: /home/yee/anaconda3/envs/robot/bin/python (found version "3.7.7") 
-- Using Python nosetests: /usr/bin/nosetests
-- catkin 0.7.20
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

a-z-e-r-i-l-a, 2020-07-09:
building with cmake arguments had also gave the `Segmentation fault (core dumped)` for me. What I did in the end was making another pacakge similar to the implementation of the roscpp_initializer that could initialize the node from the python side.

rhaschke, 2020-07-17:
Duplicate of #1521. See https://github.com/ros-planning/moveit/issues/1521#issuecomment-659933602
Also, during building, have you set the environment variable `ROS_PYTHON_VERSION`? Please, have a look into [REP 151](https://www.ros.org/reps/rep-0151.html) for more details.

closed 2020-07-17 by rhaschke

YijiongLin, 2021-01-16:
Hello @a-z-e-r-i-l-a ,

Did you successfully use moveit in python3 with Melodic (or Kinectic)? Could you please share the implementation with us?

All the best,

a-z-e-r-i-l-a, 2021-03-18:
@YijiongLin Using moveit no, but I made a package with Cython binding to have the same functionality for initializing roscpp packages that need node-handle. I will make it public as soon as I get a permission about it.

kishorekrish14, 2021-09-16:
Hello @a-z-e-r-i-l-a , 

Have you found a way to make moveit to work with python3, I am actually using a virtual environment which has python3 and trying to use moveit and it fails with the same error as you get. 

I also tried building moveit for python3 from source same as [https://answers.ros.org/question/326226/importerror-dynamic-module-does-not-define-module-export-function-pyinit__tf2/](url) but has no progress .

Did you solve the issue?

rhaschke, 2021-09-16:
@kishorekrish14, did you try the noetic-devel or master branch of MoveIt? Do you have the complete ROS Noetic installed in your virtual environment?? That's not the standard way to use ROS at least...

kishorekrish14, 2021-09-16:
Hi @rhaschke , 
I am using master branch of moveit, and 

- ROS Distro: [Melodic]
- OS Version: Ubuntu 18.04
- virtual environment: python 3.6.9

rhaschke, 2021-09-16:
You cannot use ROS Melodic. This is bound to python2!

a-z-e-r-i-l-a, 2021-09-18:
@kishorekrish14  @JatGwingLam 

first compile the ROS melodic workspace with the following cmake arguments:

`catkin config --cmake-args   -DPYTHON_EXECUTABLE=/usr/bin/python3   -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m   -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so -Wno-dev -DCMAKE_CXX_FLAGS=-Wno-comment -DCMAKE_BUILD_TYPE=Release`

Regarding Moveit, my usage was mainly about initializing a wrapped c++ class that uses node handle, so similar to what used [here](http://wiki.ros.org/ROS/Tutorials/Using%20a%20C%2B%2B%20class%20in%20Python#Class_with_NodeHandle) I had to create the roscpp_init class prior to initializing my python binding. However, after compiling moveit with the above cmake arguments in ROS Melodic, initializing roscpp_init in python was not working(`Segmentation fault (core dumped).). So I made my own implementation of `roscpp_init` using Cython that is in the following repo:
`
https://github.com/a-z-e-r-i-l-a/py_roscpp_init.git

So instead of `roscpp_init('node_name', [])` that is from moveit, simply run 

`roscpp_init.PyRosCppInit(node_name=b'py_cpp_class', n_async_spinner=2)
`
This solved my problem for running any ROS melodic c++ class in python3.