moveit/moveit #2167: termination_condition not working for all OMPL planners

pbeeson, 2020-06-25:
### Description

I'm trying to call LazyPRMstar with termination_condition: ExactSolution and I've noticed that it still runs for the full planning time even when an answer has been found.

Unsure if this is just something that is not implemented in OMPL, but the documentation seems to indicate that it should work.

### Your environment
* ROS Distro: [Melodic]
* OS Version: e.g. Ubuntu 18.04
* Source or Binary build?
Source (master for MoveIt and moveit_msgs) and compiled against OMPL 1.5.0

### Steps to reproduce
Add `termination_condition: ExactSolution` to the ompl_planner.yaml file for LazyPRMstar and try planning to a location.  It takes the full amount of planning time (5 second default for MoveIt in RViz).

### Expected behaviour
I would expect the planner to return as soon as a solution is found, as per the documentation here: https://ros-planning.github.io/moveit_tutorials/doc/ompl_interface/ompl_interface_tutorial.html#ompl-planner-termination-conditions

### Actual behaviour
It continues to run for the full time.

pbeeson, 2020-06-25:
Note it is also the case that Iteration and CostConvergence do not seem to work either.  Perhaps the PRM* algorithms in OMPL don't support this?  If so, the documentation should probably indicate this.

pbeeson, 2020-06-25:
I can also state that termination_condition does not preempt RRT* to stop prior to the planning timeout (as per the documentation that explicitly uses RRT* as an example).  Furthermore, while termination_condition is ExactSolution, it has no effect, but when I use the documented example below, after the timeout, move_group crashes.
```RRTstarkConfigDefault:
    type: geometric::RRTstar
    termination_condition: CostConvergence[10,.1]
    range: 0.0
    goal_bias: 0.05
    delay_collision_checking: 1```

pbeeson, 2020-06-26:
I see.  The "star" version of PRM explicitly keep the optimization cost at 0, even when ExactSolution is provided.   This means PRM* and LazyPRM* can never be stopped prior to the timer when a solution is found.  So, does either ApproximateSolution need to be added an OMPL condition or perhaps when termination condition is set to ExactSolution for certain non-optimizing solvers, perhaps the optimization function could be forced to be infinity?  The first a new OMPL enhancement.  The second could be a change in MoveIt (by adding an optimization function that return infinity to the problem definition when ExactSolution is set for certain planner types).  I don't really know which way people would want this handled.

zhanghengsjtu, 2022-04-09:
Hi, Pbeeson. I had a similar problem as you. In moveit (Ubuntu 16.04, ROS Kinetic) the program runs out of time when the planner is set to PRM. For comparison, I also compiled OMPL (version 1.5.0) from source code. I found that the PRM algorithm in OMPL can quickly return a valid result.
From the source code of [OMPL](https://ompl.kavrakilab.org/classompl_1_1geometric_1_1PRM.html#a5db629bb4a2bf02fb76a4cabbc5e0dcb), I found that when PRM is initialized, there is a parameter named `starStrategy` that controls whether all the time is used up. By default `starStrategy` is false,  it indicates whether the default connection strategy is the Star strategy.  When it is set to true, the planner could give a valid path quickly.
As for `termination_condition`, I tested `ExactSolution` on ROS Kinetic, it doesn't work. For option `CostConvergence`, the wiki says that it is only available in OMPL 1.5.0 and newer. For comparison, I involked API of the termination condition in [OMPL](https://ompl.kavrakilab.org/plannerTerminationConditions.html), it works well. For your reference.

liquidmetal9015, 2023-01-26:
The behavior that I get using RRTStar is as follows with respect to the termination condition in ompl_parameters.yaml:

- Leave termination_condition blank - planner runs until time runs out in all tested cases
- Write a misspelling of a termination condition in termination_condition - planner instantly fails, move group spits out an error stating that it's not a real termination condition
- Use ExactSolution - Planner runs until time runs out in all tested cases
- Use Iteration[n] - Planner fails instantly no matter what n is and reports that no work was done and a path was not found.
- CostConvergence[a,b] - planner runs until time runs out in all tested cases will all tested parameters, I've tried quite a number of different parameters for a and b.

Using ROS noetic on Ubuntu 20.04, current noetic-devel branch of moveit, apt installed ompl which is listed as 1.5.2 in it's package.xml

I'm also setting the number of planning attempts to 1.

Is there a planner where we know that the termination conditions work correctly or is it that if you are experiencing this bug then termination conditions work for none of the planners?

simonschmeisser, 2023-01-26:
See https://github.com/ompl/ompl/pull/961 for a fix for CostConvergence termination condition. It's part of ompl 1.6.0 which was released only recently and likely not yet(?) synced to noetic.

https://github.com/ompl/ompl/pull/906 fixed Iteration

In my experience I had to set the planning time absurdly high to see either cost convergence or iteration come into effect. RRTStar does implement the necessary hooks, which many other planners don't but does not report intermediate solutions that often.

I also noticed that with MoveIt's parallel planning often the solution that triggered the cost convergence was in a quite suboptimal local minimum while there existed other superior solutions in other threads.

See also https://github.com/ompl/ompl/issues/907 for my notes