moveit/moveit #2168: Running a LazyPRM plan immediately causes move_group process to die

pbeeson, 2020-06-25:
### Description

I am running today's master of moveit and moveit_msgs with OMPL 1.5.0.  If I start up a brand new moveIt config, and move the marker, I can plan with PRM, LazyPRM*, RRT, etc.  But if I choose LazyPRM, move_group immediately dies.  I tried multiple range values, and nothing seems to work.

### Your environment
* ROS Distro: [Melodic]
* OS Version: e.g. Ubuntu 18.04
* Source or Binary build?
source, master (June 25, 2020), OMPL 1.5.0

### Steps to reproduce
In description.

pbeeson, 2020-06-26:
```Thread 18 "move_group" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffa128e700 (LWP 21884)]
ompl::geometric::KBoundedStrategy<void*>::operator() (m=@0x7fffa128db58: 0x7fff940086f0, 
    this=0x7fff9000c5d0)
    at /home/pbeeson/sb/modelone_testing/thirdparty/ompl-1.5.0/src/ompl/geometric/planners/prm/ConnectionStrategy.h:180
180	                KStrategy<Milestone>::nn_->nearestK(m, KStrategy<Milestone>::k_, result);
(gdb) bt
#0  ompl::geometric::KBoundedStrategy<void*>::operator() (m=@0x7fffa128db58: 0x7fff940086f0, 
    this=0x7fff9000c5d0)
    at /home/pbeeson/sb/modelone_testing/thirdparty/ompl-1.5.0/src/ompl/geometric/planners/prm/ConnectionStrategy.h:180
#1  std::_Function_handler<std::vector<void*, std::allocator<void*> > const& (void*), ompl::geometric::KBoundedStrategy<void*> >::_M_invoke(std::_Any_data const&, void*&&) (__functor=..., 
    __args#0=@0x7fffa128db58: 0x7fff940086f0) at /usr/include/c++/7/bits/std_function.h:302
#2  0x00007fffa9708d51 in std::function<std::vector<void*, std::allocator<void*> > const& (void*)>::operator()(void*) const (__args#0=<optimized out>, this=0x7fff9000a278)
    at /usr/include/c++/7/bits/std_function.h:706
#3  ompl::geometric::LazyPRM::addMilestone (this=this@entry=0x7fff9000a190, state=0x7fff94002bf0)
    at /home/pbeeson/sb/modelone_testing/thirdparty/ompl-1.5.0/src/ompl/geometric/planners/prm/src/LazyPRM.cpp:286
#4  0x00007fffa970b98b in ompl::geometric::LazyPRM::solve (this=0x7fff9000a190, ptc=...)
    at /home/pbeeson/sb/modelone_testing/thirdparty/ompl-1.5.0/src/ompl/geometric/planners/prm/src/LazyPRM.cpp:315
#5  0x00007fffa98eec0a in ompl::tools::ParallelPlan::solveMore (this=0x7fff90007740, 
    planner=0x7fff9000a190, minSolCount=1, maxSolCount=10, ptc=0x7fffa37faf28)
    at /home/pbeeson/sb/modelone_testing/thirdparty/ompl-1.5.0/src/ompl/tools/multiplan/src/ParallelPlan.cpp:168
#6  0x00007fffeaeec6df in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007fffeb3c46db in start_thread (arg=0x7fffa128e700) at pthread_create.c:463
#8  0x00007fffea5a988f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

pbeeson, 2020-06-26:
Looks like maybe the issue is in OMPL itself.  I'll run a few more tests and paste any other findings before moving on.

pbeeson, 2020-06-26:
Seems to deterministically fail at ompl-1.5.0/src/ompl/geometric/planners/prm/ConnectionStrategy.h:180

I realize that this in an OMPL issue, but I expect it could be related to how MoveIt is setting up the problem.  Unsure if I should close this and move to OMPL.  Presumably @mamoll will see it either way?

v4hn, 2020-06-26:
If you can reproduce this with an example problem outside MoveIt, please move this to OMPL.
Otherwise it makes sense to track this here too.

pbeeson, 2020-06-26:
I think I've narrowed this down to an OMPL error.  Will file an Issue and/or PR with OMPL.

closed 2020-06-26 by pbeeson

pbeeson, 2020-06-26:
https://github.com/ompl/ompl/pull/735

pbeeson, 2020-06-26:
A temporary fix in MoveIT is to either not use a range, or to set the range AFTER max_nearest_neighbors in your ompl_planner.yaml file.