moveit/moveit #2170: Include What You Use

v4hn, 2020-06-26:
For fun and profit I just ran [include-what-you-use](https://include-what-you-use.org/) on our code base and put the results [here](https://tams.informatik.uni-hamburg.de/people/goerner/files/iwyu-moveit.log). I wanted to make it a gist, but it seem 23k lines is too much for that :)

Please have a look and tell me what you think of incorporating these changes (partially?) into MoveIt.

The idea is
1. to reduce compile time by dropping unnecessary (or unnecessarily general) includes and
2. improve correctness by not relying on transitive includes

@JafarAbdi @henningkayser @tylerjw @rhaschke and whoever else feels responsible :-)

henningkayser, 2020-06-27:
Nice! I really like all the added forward declarations and that it checks for transitive MoveIt includes. All those extra Eigen/boost/std includes seem a little excessive, though. What's the performance implication of that? I think it would be really helpful to get a less-restrictive check added to moveit_ci.

v4hn, 2020-06-27:
> I really like all the added forward declarations and that it checks for transitive MoveIt includes.

Same here. That's what made me look into it.

> All those extra Eigen/boost/std includes seem a little excessive, though.

They are, in theory, not excessive but just explicit. At the moment all of them are probably included transitively, but directly used in the respective sources.

> What's the performance implication of that?

Not sure. I did not look into automatically applying the changes.
I just wanted to get the discussion started here and maybe Jafar (or anyone else really) could have a look once some maintainers gave their opinion about this.
After all it is also a general question if we, e.g., are willing to go away from `Eigen/Dense` includes in favor of deep includes for all the things we actually use.

> I think it would be really helpful to get a less-restrictive check added to moveit_ci.

This would likely involve a lot of NOLINT pragmas to avoid excessive complains about petty stuff. I don't see that for now.

rhaschke, 2020-07-01:
In general, I like the idea to check (and auto-fix) our includes in this fashion. But I agree with Henning, that deep includes for boost or Eigen stuff is counter-productive. These meta includes were introduced for convenience and I remember that you added a similar meta include for MTC recently (stages.h?).
I'm curious how the tool finds missing includes...

v4hn, 2020-07-02:
> In general, I like the idea to check (and auto-fix) our includes in this fashion.

As I said, automating this might be a lot of work with unstable results. So I don't think that's a good project for now.

> But I agree with Henning, that deep includes for boost or Eigen stuff is counter-productive.

Actually, the reason boost *has* all these individual includes is because it always adds a lot of overhead
to rely on their convenience includes. For eigen I agree, we should refrain from including anything in `Eigen/src`.
Their documentation [explicitly describes all expected modules/header entry points](https://eigen.tuxfamily.org/dox/group__QuickRefPage.html).
Still, it might be interesting to see this benchmarked.

> These meta includes were introduced for convenience and I remember that you added a similar meta include for MTC recently (stages.h?).

Yes, but they are for convenience of *end-users*, not for the convenience of framework programmers interested in compile time. :)

> I'm curious how the tool finds missing includes...

As far as I know it looks for minimal sets of symbols provided by transitively included headers and tries to cut additional includes in favor of forward declarations.