moveit/moveit #2175: JogArm -> Servo and API migration giude

tylerjw, 2020-06-29:
I'm not sure where to put this but here are my notes about the recent changes in `jog_arm/servo`:

https://gist.github.com/tylerjw/e6443adaff45490f35932fd280af10a5

rhaschke, 2020-06-29:
As we didn't release jog_arm before, we don't need to advertise this guide, do we?

tylerjw, 2020-06-29:
Yes, and no.  People may have been using jog_arm from master and would be surprised their code doesn't build any more.

v4hn, 2020-06-29:
> People may have been using jog_arm from master and would be surprised their code doesn't build any more.

:+1: There was a lot of advertisement for the old name, so some notes somewhere make sense. Maybe just create a discourse post? You could also add a quick mention in MIGRATION.md even if it's no breaking change between distributions.

tylerjw, 2020-06-30:
I'll take a second pass at making this guide more complete and then I'll post it to discourse today.

closed 2020-09-04 by tylerjw