moveit/moveit #2176: No header for OMPLPlannerManager

gchabert, 2020-06-30:
Maybe a newbie question.
I'm trying to develop a plugin which brings some modifications to the OMPL planner. 
I would like my planner manager to inherit from OMPLPlannerManager but there is no header for this class.
Is there any reason why? If not, could it be possible to create such header?

Gilles

v4hn, 2020-06-30:
> Is there any reason why?

Probably nobody expected someone to inherit from it.

> If not, could it be possible to create such header?

Sure, feel free to create a pull-request. Make sure to test the plugin you develop on top of it before. Off the top of my head, I'm not sure pluginlib plugins can inherit from other plugins.

gchabert, 2020-06-30:
Thanks for the quick answer, I will make a pull request.
I will not try to make the enter plugin inherit, just the c++ classes inside.

closed 2020-06-30 by gchabert