moveit/moveit #2179: Unable to run demo_gazebo.launch as created by moveit_setup_assistant

thejose5, 2020-06-30:
The moveit_setup_assistant generates a demo_gazebo.launch file which is supposed to connect moveit with a robot simulated in Gazebo. However, this launch file cannot be used directly without first making significant changes to files in the moveit_config package generated by the setup assistant.

Based on the number of questions on online forums and external blog posts I have seen about connecting moveit to a gazebo simulation, it seems clear to me that this is common knowledge. 

I was wondering why the setup assistant is built such that it does not generate the demo_gazebo.launch in a directly usable way. Moreover, can a change be made in the setup assistant to allow it to generate a demo_gazebo.launch file that can be run without any additional changes?

k-ish0212, 2020-07-14:
I had a same problem and it was helpful for me. https://medium.com/@tahsincankose/custom-manipulator-simulation-in-gazebo-and-motion-planning-with-moveit-c017eef1ea90

thejose5, 2020-07-17:
Thank you for your reply. I have seen the blog post that you have linked and I've used it too. I just wanted to know if the developers created the demo_gazebo.launch file like this on purpose and whether it can be fixed.

dhiegomaga, 2020-08-06:
I agree, it's such a simple thing, you wanna be able to simulate the robot in Gazebo, yet its incredibly hard to fix all the poorly implemented code. 

I had the same issue here https://github.com/ros-planning/moveit/issues/2237

davidhjp, 2020-11-14:
I also hope this get fixed soon.

rhaschke, 2022-10-05:
This was addressed in various commits, e.g. #2932, #2945, #2946

closed 2022-10-05 by rhaschke