moveit/moveit #2183: spline optimizer does weird things

pbeeson, 2020-07-02:
### Description

Reporting since the docs say you are looking for feedback on this optimizer.

I was looking to use the Spline Optimizer on a trajectory.  This is a VERY simple test case where a 6-DOF robot spins its first joint from -2pi to +2pi.  The initial trajectory is something like ~360 points.  If I use the old ParabolicTime optimizer with its defaults, I get ~6 seconds, which of course is wrong since the real answer (Assuming infinite jerk) is 4.09 s.  If I crank way up the number of iterations and crank way down the max distance between waypoints, I get out ~2.1 seconds, so its treating acceleration as infinite.  Thus, wanting to use the newer Spline Optimizer.

Using the spline optimizer on this very simple problem, I get 23 second trajectory time.  Which is just so far away from realtity.  I also get REALLY weird graphs for accelerations and velocity, though position is a straight line.  I will attach those in the comments.

Long story short, I think either that implementation is bugged or it's not able to handle "realistic" trajectories of many points.

### Your environment
* ROS Distro: [Melodic]
* OS Version:  Ubuntu 18.04
* Source or Binary build?
Source (master)

### Steps to reproduce
Presumably any robot can be set up to have 6 r/s vel and 3 r/s/s on the base roll and then move it between its extremes.

pbeeson, 2020-07-02:
FYI, the time_optimal_trajectory_generation works out of the box, and provides a overall time of 4.09s exactly.

henningkayser, 2020-07-03:
> FYI, the time_optimal_trajectory_generation works out of the box, and provides a overall time of 4.09s exactly.

I just wanted to mention this, TOTG should work in most cases.

The problems you described with the other parameterization methods have already been discussed in other issues some time ago (i.e. https://github.com/ros-planning/moveit/issues/160, https://github.com/ros-planning/moveit_core/issues/167). However, these should not be as common in "realistic" scenarios as they have been used successfully on many robots and applications with minimal issues (typical would be jerks, oscillation). The extreme problems indicate to me that there is something wrong in the velocity/acceleration limits in your setup. Did you plot the solution trajectories or the executed joint states? I would still say that TOTG should be used as a default, even though this method is also by no means perfect.

v4hn, 2020-07-03:
for clarification, you are talking about [IterativeSplineParameterization](https://github.com/ros-planning/moveit/blob/master/moveit_core/trajectory_processing/include/moveit/trajectory_processing/iterative_spline_parameterization.h), right?

> I will attach those in the comments.

Could you attach the plots please? I guess the usual recommendation these days is to rely on TOTG for anything where good dynamics are relevant, but the other types should still provide reasonable results.

@henningkayser How about a pull-request to make TOTG the default parameterization in the master branch? We can still change it in for noetic.

pbeeson, 2020-07-03:
> However, these should not be as common in "realistic" scenarios as they have been used successfully on many robots and applications with minimal issues (typical would be jerks, oscillation). The [...]

There is nothing wrong with the velocity acceleration values.  All velocities were 6 and all accelerations were 3.  Only one joint moved, the other 5 stayed at the same values.

The trajectory I used was generated from ompl and smoothed to 2 points (It is just one joint moving from -360 to +360).  Then got upsampled to about 375 points in the trajectory due to my ompl max fraction value being 0.0005.

Without changing anything else, But the one line of which optimizer to use, I switched to TOTG and it solved perfectly.  

So the spline method really failed quite hard on a 2 waypoint, one joint moving (5 joints with zero motion) Trajectory on 375 points. My guess is the spline method failed due to the density. This is a “Realistic” scenario to me.  Moving the base roll by 360 degrees or more is quite common in calibration or to zero out joints after rollout.

I did make plots that I share.  They weren’t great screen grabs my me, But I don’t want to go back and make new ones at this point since I’ve moved on.

> I would still say that TOTG should be used as a default, even though this method is also by no means perfect.

This should probably be documented then.  The Moveit docs seem to have tacked on TOTG as an afterthought.  I was under the impression from reading through docs and issues that the spline approach was being tested to replace the parameterizarion approach.

FYI, I moved to TOTG but it also had issues, getting bogged down on trajectories of thousands of points.  I submitted a minor PR last night that helps get around this.

pbeeson, 2020-07-03:
> for clarification, you are talking about [IterativeSplineParameterization](https://github.com/ros-planning/moveit/blob/master/moveit_core/trajectory_processing/include/moveit/trajectory_processing/iterative_spline_parameterization.h), right?
> 
> > I will attach those in the comments.
> 

Yes.  Will send the plots in a few hours.

> 
> @henningkayser How about a pull-request to make TOTG the default parameterization in the master branch? We can still change it in for noetic.

I would vote for this, but please note that I found a scenario with a 3000 point trajectory that did a complicated motion where TOTG bogged down and took 15 seconds to solve.  It had an good answer but that’s too long.  PR #2185 was meant to get around this but if you make TOTG the default that new param in the PR should probably be something bigger than 0.001 for most users.

pbeeson, 2020-07-03:
[image]

Position plot of a 375 point, one joint trajectory moving from -360 to +360 degrees.  All other joints don’t move.

[image]

Velocity plot after running iterative Spline optimizer when velocity of all joints was set to 6.  Notice it gets nowhere near 6.  TOTG makes a nice line up from 0 to 6 then plateaus, then line back down to 0.

[image]

Acceleration plot after running spline optimizer.  While TOTG is a step function from zero to 3 to zero to -3 this is just craziness.

v4hn, 2020-07-03:
These plots show totally broken behavior, hopefully only due to some bug.
The velocity profile should at least *approach* the maximum velocity in your case.

> I would vote for this, but please note that I found a scenario with a 3000 point trajectory that did a complicated motion where TOTG bogged down and took 15 seconds to solve.
> It had an good answer [...]

Yes, that kind of reasoning is why we still have IPTP as the default. It's bad, but it's also fast.
That was one of the reasons for [this issue](https://github.com/ros-planning/moveit/issues/2110).
Notice though that "default" is not really correct, because users cannot reconfigure it easily for all interfaces.
Instead, for some trajectories you have to rerun the parameterization you want afterwards.

Aside from that, I would argue that if I ask for a parameterization of a trajectory with 3000 points, I would not expect a heavily down-sampled result. If you want a down-sampled trajectory, you could do that *before* the parameterization.
On the other hand, if you *require* such a high resolution, then you also should be willing to spend computation time on it.

pbeeson, 2020-07-03:
FYI, the velocity plot is the correct one.  The acceleration plot I see from the x axis is not from the same run. I'm regenerating these plots now for posterity, and will edit them above them probably delete this comment.

pbeeson, 2020-07-03:
> On the other hand, if you _require_ such a high resolution, then you also should be willing to spend computation time on it.

Let me respectfully disagree with you here.  As someone that has been in the robotics research community for over 20 years, and put robots into high-volume production environments, this is not an acceptable answer.  15 seconds to figure out the time parameterization of a 3000 point trajectory should not be considered acceptable by any of us.

Prior to using MoveIt with my current position, I developed my own alternative motion planning framework with my prior company, and we used TOPP-RA.  It would solve for 10s of thousands of very dense points for sub-mm precision trajectories in a matter of milliseconds.  So I know for a fact that this can be done.  If I had the time and resources, I'd probably be using TOPPRA, but my current goal is to get a baseline system up and running ASAP before making it more robust long term.

gavanderhoorn, 2020-07-03:
>  If I had the time and resources, I'd probably be using TOPPRA

There is a service for that: https://github.com/ros-planning/moveit/issues/2110#issuecomment-634058198.

And tracking TOPP-RA integration in #2110.

pbeeson, 2020-07-03:
Sorry @v4hn I missed your pointer to TOPP-RA discussed in your earlier comment.  It's a holiday here and I'm not supposed to be working, but I was trying to these plots uploaded on the sly when my wife wasn't looking.

That, I've used TOPP-RA for the past 2 years, and it solved in milliseconds (unsure exactly why Dave was saying it was slow), but it did take some pre-processing of the data to get correct solutions in many cases due to numerical instabilities in the Python code.  I'm hoping the C++ port cleans this up, but I'm sure that will have some growing pains too.

Thanks for the help.  I'm inheriting a MoveIt setup in a new position, and getting back to it for the first time 5 years.  Unfortunately, there haven't been magic solutions to these hard problems in that time.

v4hn, 2020-07-03:
> on the sly when my wife wasn't looking

Talk to me, I have my kid crying in the background quite often. :wink:

>  this is not an acceptable answer.

Acknowledged and mostly agreed.

> 15 seconds to figure out the time parameterization of a 3000 point trajectory should not be considered acceptable by any of us.

I was under the impression that the plugin validates newly created waypoints in terms of collision checking, but looking at the code this is not the case. For plain dynamics parameterization 15 seconds is indeed frustrating.

My actual concern remains though: It seems unexpected to have a time parameterization adapter down-sample a high-resolution trajectory for performance reasons. If you don't need the resolution, why not generate the trajectory with lower resolution to begin with? But If you do need it, you have to upsample it after the parameterization again...

A TOPP-RA adapter should have some priority in MoveIt, but nobody who actually gets stuck on issues with the existing parameterizations was willing to contribute on it :confused: 

> Prior to using MoveIt with my current position, I developed my own alternative motion planning framework with my prior company, and we used TOPP-RA. It would solve for 10s of thousands of very dense [...]

In that sense thanks for your current stream of contributions to the project. It's much appreciated.

felixvd, 2020-10-21:
I took some example videos with a UR5e today. For the longest time I thought that the jerky movements I sometimes see were network lag spikes, but it actually seems like they are due to the time parameterization. You can download and investigate the trajectories yourself (shoutout to [this rqt plugin](http://wiki.ros.org/rqt_joint_trajectory_plot)).

The oscillating effect is clearest for the downward motion and IPTP. TOTG looks a little better, but I don't know that the graph inspires more confidence. I ran out of time and couldn't test TOPP, but I might be able to update this post with it at some point.

## IPTP

http://www.youtube.com/watch?v=RSZEKRfi0Bc

Example graph (when moving downward)
[image]

[Full trajectory (moving downward)](https://github.com/ros-planning/moveit/files/5416724/downward_motion_iptp.zip)

## TOTG

http://www.youtube.com/watch?v=I_2aPcvBIhE

Example graph (when moving downward) (using resample_dt = 0.01 instead of the default 0.1)
[image]

[Full trajectory (moving downward)](https://github.com/ros-planning/moveit/files/5416710/downward_motion_totg.zip)

gavanderhoorn, 2020-10-21:
Did you not test the spline-based one for a reason? We use it with our KR210 sometimes and it doesn't *sound* too bad (you can actually hear it pretty easily when a trajectory isn't smooth with KUKAs and RSI).

felixvd, 2020-10-22:
Only because IPTP was the default in the [cartesian path planning service](https://github.com/ros-planning/moveit/blob/master/moveit_ros/move_group/src/default_capabilities/cartesian_path_service_capability.cpp#L162) and I didn't have the time. I updated the last post with ISP and TOPP-RA.

AndyZe, 2020-10-22:
I'm interested in this because PickNik has a similar product (except it smooths one level further - to jerk).  https://picknik.ai/products/tracksuite/

If the algorithms were truly time-optimal, either velocity or acceleration would be maxed at every time step. Those limits for shoulder_pan_joint of the UR5e (that Felix tested with) are (vel limit = 3.14, accel limit = 3.14). https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/ur5_e_moveit_config/config/joint_limits.yaml

None of the acceleration or velocity values get close to 3.14, so I would say it's not even close to time-optimal and there is potential for much jerkier motion before limits are hit! Maybe one of the other joints was the limiting factor when the DOF were synchronized.

AndyZe, 2020-10-22:
@felixvd Are you only moving one joint? That would be the best test case. Otherwise we can't tell if joint synchronization is to blame.

felixvd, 2020-10-23:
No, it's a cartesian motion as shown in the video. You can download the zip files and publish the trajectories to regenerate the graphs and check the other joints, but they aren't moving very fast either.

> I'm interested in this because PickNik has a similar product (except it smooths one level further - to jerk). https://picknik.ai/products/tracksuite/

I figured that's why PickNik would be more interested in selling that than fixing this ;)