moveit/moveit #2208: Point cloud topic has data, but NO Octomap in Rviz!

yiyinglai, 2020-07-14:
### Description
On hardware, point clouds are provided by RealSense D415. PointCloud2 exists, but no Octomap appears.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04

### Expected behaviour
Octomap shows up as in simulation.

### Actual behaviour
No Octomap in Rviz.

### Setup
Content of `niryo_one_moveit_config/config/sensors_3d.yaml` :

    sensors:
      - filtered_cloud_topic: filtered_cloud
        max_range: 5.0
        max_update_rate: 1.0
        padding_offset: 0.1
        padding_scale: 1.0
        point_cloud_topic: /camera/depth/points
        point_subsample: 1
        sensor_plugin: occupancy_map_monitor/PointCloudOctomapUpdater

Content of `niryo_one_moveit_config/launch/sensor_manager.launch.xml`:

    <launch>
    
      
    
      
      <rosparam command="load" file="$(find niryo_one_moveit_config)/config/sensors_3d.yaml" />
    
      
      
      <param name="octomap_resolution" type="double" value="0.025" />
      <param name="max_range" type="double" value="5.0" />
    
      
      <arg name="moveit_sensor_manager" default="niryo_one" />
      <include file="$(find niryo_one_moveit_config)/launch/$(arg moveit_sensor_manager)_moveit_sensor_manager.launch.xml" />
    
    </launch>

### Simulation
It works for me, as shown in the following figure (I used [RealSense plugin](https://github.com/pal-robotics/realsense_gazebo_plugin/tree/kinetic-devel) to simulate the camera).

### Excluded reasons
I have `ros-kinetic-octomap` and `octomap_rviz_plugins` installed.
There is data in the point cloud topic.
The config file has the correct point cloud topic name.

simonschmeisser, 2020-07-16:
Since it seems to work with simulated data you need to check what's different between your simulated data and the real data. Candidates are:

- tf (where is the point cloud located)
- topics (is the octomap updater listening to the correct topic)
- message format

Please also note that support questions should be asked on answers.ros.org

If there is actually a bug please provide a github repo with the configuration and a rosbag with sensor data as a minimally working example (MWE)

ps: please try keep your issue titles "calmer"

yiyinglai, 2020-07-16:
@simonschmeisser First off, my apology for the exclamation marks and asking questions here. In my experience, answer.ros.org has not been useful (I have asked several questions there, some for discussion, some for a specific problem, and got zero answer from other users).

Thanks for your suggestion. I don't think they are the real reasons. Under the secontion **Alternative?**, I explicitly launched the `octmap_server_node` and if any of those three candidates is the cause, there wouldn't be an Octomap in Rviz, right?

The topic names are definetly correct. I changed the config before switching between simulation and hardware and verified the publication/subscription relationship using `rqt_graph`.
Point cloud topic `/camera/depth/points` goes to `/move_group` in the simulation node graph:
[image]
Point cloud topic `/camera/depth/color/points` goes to `/move_group` in hardware node graph:
[image]

And message format is also the same. Again, when explicitly launching `octomap_server_node`, it worked.

yiyinglai, 2020-07-23:
I made a [repo](https://github.com/yiyinglai/mwe-ros-planning-moveit-issues-2208) of a mwe. Two rosbags are provided, one from hardware, one from Gazebo simulator.

When the rosbag file from Gazeb simulator was played, the Octomap showed up.
[image]

When the rosbag file from hardware was played, no Octomap is present and the following showed up:
` Transform error: "right_gear" passed to lookupTransform argument source_frame does not exist.`

I thought the tf complaint is not relevant because `right_gear` has nothing to do with camera frames, as shown in [frames.pdf](https://github.com/ros-planning/moveit/files/4963811/frames.pdf). 

Is it because the tf (even though it seems `right_gear` or any other child links of `mainsupport` will not affect `camera_bottom_screw_frame` and its children)?

simonschmeisser, 2020-07-23:
I guess you need all transforms defined for self-filtering to work? Best have a look at the code yourself here https://github.com/ros-planning/moveit/blob/master/moveit_ros/perception/pointcloud_octomap_updater/src/pointcloud_octomap_updater.cpp

yiyinglai, 2020-07-28:
With all the transforms available, the Octomap is up now. Thanks!

closed 2020-07-28 by yiyinglai