moveit/moveit #2209: Configuration parameter "subspaces" in the OMPL interface undocumented

JeroenDM, 2020-07-15:
### Description

MoveIt's OMPL interface uses a parameter "subspaces" in the [PlanningContextManager](https://github.com/ros-planning/moveit/blob/091bdfb765e46ba399e6983f49a81c502b0bcb65/moveit_planners/ompl/ompl_interface/src/planning_context_manager.cpp#L350). This modifies an attribute [ std::vector<ModelBasedStateSpacePtr> subspaces_](https://github.com/ros-planning/moveit/blob/9dcc93cf96b8910698a0848be9799c04a644e2af/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/model_based_planning_context.h#L71) of the struct `ModelBasedPlanningContextSpecification`.

I cannot find any documentation or usage of this in the rest of the codebase or tutorials. What does this parameter do? Am I looking in the wrong places?

I'm asking this in the context of #2092.

### Your environment
* ROS Distro: [Melodic]
* OS Version: e.g. Ubuntu 18.04
* Source build of master branch

v4hn, 2020-07-15:
Sorry for not giving an anwers, but if once you understand it, please document it somewhere.
It seems to be about some feature of OMPL planners that I've never heard about (and that might not even exist these days...).

JeroenDM, 2020-07-15:
Ah, indeed, it could be meant to provide support for OMPL's [CompoundStateSpace](https://ompl.kavrakilab.org/classompl_1_1base_1_1CompoundStateSpace.html), where the terminology "subspace" is used throughout the codebase.

But as far as I can tell, it is an unfinished implementation and the code is never used. Is there a more systematic way to check this other than just searching for the use of the `subspaces_` attribute?

JeroenDM, 2020-07-15:
I ran the tests for `moveit_planners_ompl` and `moveit_ros_planning_interface` with the [subspaces code removed](https://github.com/JeroenDM/moveit/commit/611689db82b04cd8b8bb133e26ca1e9cb7ddbcff) using:
```
catkin run_tests --no-deps --this
```
in their directory and did not get any failing tests. But of course I'm only wildly guessing what the code is meant for.