moveit/moveit #2210: Setup Assistant demo_gazebo.launch broken?

kartner20, 2020-07-16:
### Description
After launching `demo_gazebo.launch` and using the Rviz planning tool, Execute will result in the following error:
```
[ERROR] [/move_group] [1594939268.772654757, 13.622000000]: Unable to identify any set of controllers that can actuate the specified joints: [joint0, joint1, joint2, joint3, joint4, joint5, joint6]
[ERROR] [/move_group] [1594939268.772702623, 13.622000000]: Known controllers and their joints:
[ERROR] [/move_group] [1594939268.772758737, 13.622000000]: Apparently trajectory initialization failed
```
### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Binary
* apt list shows: 1.0.4-1bionic.20200630.155528
### Steps to reproduce
Followed the setup assistant tutorial, then launched demo_gazebo. No errors or warnings till you execute a plan from the RViz motion planning panel.
### Expected behaviour
The arm should move in Rviz and Gazebo.
### Actual behaviour
Arm only moves in Rviz.
### Backtrace or Console output
above
In `ros_controllers.yaml` should `controller_list` be empty?

kartner20, 2020-07-16:
@jafarabdi It looks like you did a lot of the gazebo file generation?

dhiegomaga, 2020-08-06:
Yeah, I had the same issue here
https://github.com/ros-planning/moveit/issues/2237

davetcoleman, 2020-11-06:
Can we close this issue in favor of #2237 ?

simonbogh, 2021-02-09:
@kodie-artner no, `controller_list` should not be empty. Check my answer here and see if it helps you: https://github.com/ros-planning/moveit/issues/2237#issuecomment-776155882

closed 2021-02-12 by kodie-artner