moveit/moveit #2212: New libmesa shipped in Bionic causes a regression in mesh_filter tests

rhaschke, 2020-07-17:
As already pointed out in #2180, a newer version of libmesa (in Ubuntu 20.04) causes a new regression of the mesh_filter tests.
This newer, but broken version of libmesa 20.x [seems to be rolled out in Ubuntu 18.04 now as well](https://packages.ubuntu.com/bionic-updates/libgl1-mesa-dev), introducing Travis failures from d6666a9aee734acd1891a43657cf162c9ca4e9b7 ([working](https://travis-ci.com/github/ros-planning/moveit/builds/174473351)) to edc9cf6e55ffcd2f3dbe18f2eb8602d913fdde48 ([failing](https://travis-ci.com/github/ros-planning/moveit/builds/176025591)).
The fix proposed in #2180, increases the accuracy threshold for some tests. I [retriggered one (out of 3) Travis jobs](https://travis-ci.com/github/ros-planning/moveit/jobs/357433320) of this PR to validate that the fix also applies for Ubuntu 18.04.
@henningkayser or @v4hn, please consider to approve and merge #2180 to fix this issue.

jschleicher, 2020-07-24:
Fixed in #2180, so close here?

rhaschke, 2020-07-25:
Indeed.

closed 2020-07-25 by rhaschke