moveit/moveit #2213: META: Renaming "master" to "main"

felixvd, 2020-07-17:
Sooner or later [this topic](https://discourse.ros.org/t/consider-renaming-default-branches/15385/) will come up here. We might as well summarize the discussion before we are blindsided in a ~maintainer meeting~ working group meeting and end up wasting time.

**Summary:** There is a [discussion](https://tools.ietf.org/id/draft-knodel-terminology-00.html#rfc.section.1.1.1) about the use of "master/slave" terminology in tech. The branch name "main" is considered preferable to "master". Shall we switch?

Regardless of the controversy, "main" is likely a better branch name than "master" (the [original author thinks so](https://twitter.com/xpasky/status/1271477451756056577)), particularly in our use case (it is the active dev branch, not the copy of record). I don't know that it is worth the effort of changing, but to have an idea of what it entails, I had a quick look around.

**What would need to be changed?**

Assuming we add a `main` branch and keep `master` around:

- Add a `main` branch of the `ros-planning` repositories and change their default branch setting on Github
- Change a few mentions in `MIGRATION.md`, `codecov.yaml` and on the website
- Change the references in the Dockerfiles in `moveit` and a few Travis references in `moveit_ci`
- Decide what to do about the `master-source`, `master-ci`, `master-experimental` docker images
- [Rename the branch in local clones](https://www.hanselman.com/blog/EasilyRenameYourGitDefaultBranchFromMasterToMain.aspx)

Deleting the `master` branch (or renaming it to `main`) would affect `.rosinstall` settings etc. of downstream users, so that isn't tractable without at least 12 months notice. Adding `main` as the default branch and keeping `master` on life support does not seem to be intrusive or difficult, but does require effort.

**Is it worth it?**

I don't know, I just wanted to compile the information. We have [added a code of conduct](https://github.com/ros-planning/moveit/pull/647) before to be more inclusive, but that did not come with any technical consequences, whereas this does. If I was forced to make a choice, I would probably vote "If someone wants to take care of it, go for it".

rhaschke, 2020-07-17:
I think we should postpone this discussion to the point when git decides to rename its default branch. Let's focus on real development. I never even thought of a connotation with slavery.

v4hn, 2020-07-17:
*sigh* agreed.

gavanderhoorn, 2020-07-18:
If I've learned anything the past few days (and judging my inbox I've had plenty of input/help there): @rhaschke: it doesn't matter what your association with a word is. I would recommend to read [this](https://discourse.ros.org/t/consider-renaming-default-branches/15385/36) comment by Bill Smart on the discussion @felixvd references to get an idea of the state of affairs and sensitivity of this subject.

This is about perception and emotion, not a technical discussion.

I'm still not sure I understand how this will scale, but that's not the point here.

v4hn, 2020-07-18:
> it doesn't matter what your association with a word is.

Indeed. As already Lewis Carroll made us understand in Humpty Dumpty's famous "When *I* use a word" quote.

> This is about perception and emotion, not a technical discussion.

As with all change-here-to-be-inclusive debates there is only one way forward and I am convinced no maintainer directly opposes the change as long as everything is changed consistently.

davetcoleman, 2020-08-20:
_just seeing this thread_

I've been following this trend in tech and the other threads in the ROS community and I agree with @felixvd that it is inevitable. I agree with @gavanderhoorn there are no technical merits, and I too really liked Bill Smart's [write-up](https://discourse.ros.org/t/consider-renaming-default-branches/15385/36) @wdsmart.

I propose we don't change anything in MoveIt 1 since it will soon be deprecated anyway and is a fair amount of effort, but we should do the full switch to the "main" branch naming in MoveIt 2 while there are still very few adopters. I believe Open Robotics said something to that effect also.

My personal belief: the BLM movement is very important in the US and elsewhere, and I support it. I do think there are many more significant and pressing issues than the "master" branch naming that we, as a community, need to address regarding systemic racism and diversity, [particularly in an open source community](https://www.wired.com/2017/06/diversity-open-source-even-worse-tech-overall/). I am worried that this semantic naming change is a distraction and excuse to claim progress without engaging in real issues. 

That said, let's switch to using a "main" branch with MoveIt 2 as a first step.

davetcoleman, 2020-09-14:
moveit 2 is now using "main" as its default branch:
https://github.com/ros-planning/moveit2

I think this issue can be closed.

closed 2020-09-14 by davetcoleman