moveit/moveit #2217: Can't find the moveit_jog_arm package

vinay-lanka, 2020-07-20:
### Description

Cannot find the moveit_jog_arm package either in moveit_experimental or moveit_ros.
I was trying to run the Arm Jogging in Real-Time example in the official docs.
According to [this PR](https://github.com/ros-planning/moveit/pull/2165), it was moved to moveit_ros.
If it was removed/relocated, the docs could be updated to show the latest changes.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source or Binary build? : Source

### Steps to reproduce
I tried to run the Arm Jogging in Real-Time example in the official docs. Could not locate the package. 

### Expected behaviour
spacenav_cpp was supposed to run with -  `roslaunch moveit_jog_arm spacenav_cpp.launch` 

### Actual behaviour
Trying to launch the spacenav example gives an error.

### Backtrace or Console output

`RLException: [spacenav_cpp.launch] is neither a launch file in package [moveit_jog_arm] nor is [moveit_jog_arm] a launch file name
The traceback for the exception was written to the log file`

JafarAbdi, 2020-07-20:
Hi @vinay-lanka,

This PR https://github.com/ros-planning/moveit/pull/2165 renamed the package from `moveit_jog_arm` to `moveit_servo`, so you should run `roslaunch moveit_servo spacenav_cpp.launch` -- see https://github.com/ros-planning/moveit/tree/master/moveit_ros/moveit_servo

vinay-lanka, 2020-07-20:
Ahh got it thanks! It's not pointed out in the docs, would love to make a PR to update.

closed 2020-07-20 by vinay-lanka

ZhouJankin, 2022-02-25:
Hi @JafarAbdi 
When I run "roslaunch moveit_servo spacenav_cpp.launch“ It warns that I haven't the launch file "moveit_servo".
Could you give me some advises about how can I install the moveit_servo?
I used to use the command:$sudo apt install ros-melodic-moveit