moveit/moveit #2225: moveit_config demo.launch issue after moveit_setup_assistant setting. anyone can help? thanks

HansAItec, 2020-07-27:
after ues the "moveit_setup_assistant".  a moveit_config forlder has been created here .
when i execute the code "roslaunch moveit_config demo.launch". 
report this:

RLExcepting: error loading <rosparam> tag:
        file dose not exit [3]
XML is <rosparam command=:"load" file="3" ns="arm"/>
The traceback for the exception was written to the log file

so whats the problem and how i can solve it?

ps: the system is Ubuntu18.04 ROS Melodic. I have tried the same process on Ubuntu16.04 ROS Kinetic and there is no such probem.

v4hn, 2020-07-27:
Can you please provide the name and full contents of the file that contains the broken xml line?

felixvd, 2020-09-16:
Closing for inactivity. Please reopen if your problem persists.

closed 2020-09-16 by felixvd

marcinolokk, 2021-08-02:
Just encountered the same thing. I simply misread the field "Kin. parameters file:" as "Kin. Solver Attemps" and therefore typed 3 as input.

Gaurav37, 2021-11-27:
I am getting this issue now. How did you solve this error?
Should I comment this line in moveit generated planning_context.launch?

rhaschke, 2021-11-27:
@Gaurav37, please open a new issue report and provide all the required details like error message, version information, etc.

jeongnow97, 2022-06-03:
Did you solve the problem?
I got a same error..
I think the problem is that file version is kinetic but my ubuntu version is melodic!
But i don't know how to solve this error..

rhaschke, 2022-06-03:
@jeongnow97, please open a new issue report and provide the exact error message as well as the offending launch file.

jeongnow97, 2022-06-07:
@rhaschke 

Thank you for replying :)

More details are here

I am using ubuntu 18.04
And I installed ROS package following open manupulator e-manual
https://emanual.robotis.com/docs/en/platform/openmanipulator_x/quick_start_guide/#setup
(by changing the word in command kinetic to melodic)

And open the moveit by command 
$ roslaunch moveit_setup_assistant setup_assistant.launch

Then follow the Youtube and finish the assistant. launch file was created

For the last step, I am trying to open the laucn file by using command
$ roslaunch open_manipulator_moveit demo.launch

But I encountered the error said that
Deprecated: xacro tag 'RefLink' w/o 'xacro:' xml namespace prefix (will be forbidden in Noetic)
when processing file: /home/irl-03/now_ws/src/open_manipulator/open_manipulator_description/urdf/open_manipulator.gazebo.xacro
included from: /home/irl-03/now_ws/src/open_manipulator/open_manipulator_description/urdf/open_manipulator_robot.urdf.xacro
Use the following command to fix incorrect tag usage:
find . -iname "*.xacro" | xargs sed -i 's#<\([/]\?\)\(if\|unless\|include\|arg\|property\|macro\|insert_block\)#<\1xacro:\2#g'

RLException: error loading <rosparam> tag: 
	file does not exist [3]
XML is <rosparam command="load" file="3" ns="arm"/>
The traceback for the exception was written to the log file

Now there is no progress..

I thought it is because of the version
But I haven't found the solution yet...

rhaschke, 2022-06-07:
There are multiple issues here: First, xacro issues a warning that the tag RefLink (probably a macro) is used without the `xacro:` namespace prefix. This should be fixed upstream in https://github.com/ROBOTIS-GIT/open_manipulator.git.

Second, roslaunch throws an exception because a file is missing. Unfortunately, the file name is not shown. Is the name empty?
Please provide a zip file comprising the whole `open_manipulator_moveit` package that you generated with MSA. Please do so in a _new_ issue and reference that new issue here.

jeongnow97, 2022-06-07:
@rhaschke
Thank you for replying

My coworker helped me and solved the problem

I remove the 
<rosparam command="load" ns="arm" file="3"/>
<rosparam command="load" ns="gripper" file="3"/>
in ws/src/open_manipulator_moveit/launch/planning_context.launch

It was very hard find that sentences in workspace

I don't know what is that sentences' role
But Rviz is worked and other function are worked well so far

I will do in a new issue if i found a new issue

rhaschke, 2022-06-07:
What exactly did you remove from planning_context.launch? Your previous post is incomplete.

jeongnow97, 2022-06-07:
oh sorry

    <rosparam command="load" ns="arm" file="3"/>
    <rosparam command="load" ns="gripper" file="3"/>

this one!

jeongnow97, 2022-06-08:
Oh.. I didn't check thoroughly..
Thank you so much!!