moveit/moveit #2228: Two mesh_filter_tests fail locally

jschleicher, 2020-07-28:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source  build according to https://moveit.ros.org/install/source/
* master, dc909df5d2

### Steps to reproduce
Build an run tests using `catkin run_tests`
and `catkin_test_results`

### Expected behaviour
All tests passed

### Actual behaviour
```
[  PASSED  ] 22 tests.
[  FAILED  ] 2 tests, listed below:
[  FAILED  ] float_test/MeshFilterTestFloat.float/7, where GetParam() = 3.5
[  FAILED  ] float_test/MeshFilterTestFloat.float/9, where GetParam() = 4.5

 2 FAILED TESTS
```
https://gist.github.com/jschleicher/d5142233e4e668fe3b6e85bb1aee8b30

Any ideas?

JafarAbdi, 2021-02-03:
I'm pretty sure this's related to https://github.com/ros-planning/moveit/issues/374 which is if I recall correctly due to a bug in Mesa, this can be fixed by exporting the following env variable before running the tests `export LIBGL_ALWAYS_SOFTWARE=1
`

jschleicher, 2021-02-04:
Yep, duplicate of #374 - closing here.

closed 2021-02-04 by jschleicher