moveit/moveit #2233: gui-docker: improving docs

myFree, 2020-08-03:
I would like to have a better understanding of how https://github.com/ros-planning/moveit/blob/master/.docker/gui-docker is working; and if there are any "trades-offs" or potential "gotchas" with the approach.

Does http://wiki.ros.org/docker/Tutorials/GUI include the approach that gui-docker is using? As far as I can tell; it does not; as gui-docker is storing X11 access rights in temp file to be passed into docker container; and as I read the tutorial; it does not cover that approach.

As I understand - the tradeoffs in the http://wiki.ros.org/docker/Tutorials/GUI  tutorial range from very insecure through to; may hit some runtime issues due to home directly setup through to; performance degrade.

Does the approach in gui-docker manage to have no security holes; no potential run time corner cases AND no performance trade offs?

Perhaps updating http://wiki.ros.org/docker/Tutorials/GUI with this option would be a good way to go in terms of fleshing out helpful docs to new comers to the ROS and friends eco-system? 

(I can have a go at doing that if I can get my understanding to a point that I can create an accurate update.)

v4hn, 2020-08-03:
@tfoote and @rhaschke worked on this script and are the only people who might comment on this.

rhaschke, 2020-08-03:
gui-docker follows the [isolated way](http://wiki.ros.org/docker/Tutorials/GUI#The_isolated_way) to pass X11 permissions, but it does not change the user. Thus running the docker container as root imposes some security hole. You should trust the container you are executing ;-)
The gui-docker script just provides a convenient wrapper script for most use cases we encountered so far.

closed 2020-08-03 by rhaschke

myFree, 2020-08-03:
ah; right;thanks for highlighting that for me.

tfoote, 2020-08-13:
@myFree  Since this was created I've put together rocker to try to make things a little easier and more extendable for other support than just GUI: https://github.com/osrf/rocker