moveit/moveit #2234: Melodic Perception Pipeline Tutorial loads empty rViz

myFree, 2020-08-03:
### Description

I see the image on the left; expect to see the image on the right
[image]

### Your environment
* ROS Distro: osrf/ros melodic-desktop-full 
* OS Version: Ubuntu 18.04.4 LTS

### Steps to reproduce
After following the Getting Started pages; with both repos checked out on the `melodic-devel` branch
`roslaunch moveit_tutorials detect_and_add_cylinder_collision_object_demo.launch`

### Expected behaviour
Populated Octomap appears

### Actual behaviour
rViz only displays the robot arm

https://gist.github.com/myFree/3c3a5fd662751b462838d297ea5c55e1

Any suggestions on how I triage this further on my side most welcome.

JafarAbdi, 2020-08-04:
Have you tried to run `export LIBGL_ALWAYS_SOFTWARE=1` as shown [here](https://ros-planning.github.io/moveit_tutorials/doc/perception_pipeline/perception_pipeline_tutorial.html#running-the-interface)

myFree, 2020-08-04:
Thanks for the suggestion; I had not tried with that - however the result is the same after exporting it before launching

```
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
[point_cloud_preprocessor-11] process has died [pid 3144, exit code -11, cmd /root/ws_moveit/devel/lib/moveit_tutorials/cylinder_segment __name:=point_cloud_preprocessor __log:=/home/user/.ros/log/549d9fe8-d66d-11ea-a24d-0242ac110002/point_cloud_preprocessor-11.log].
log file: /home/user/.ros/log/549d9fe8-d66d-11ea-a24d-0242ac110002/point_cloud_preprocessor-11*.log

```

myFree, 2020-08-05:
Output of `root@7e34595fbaf7:~#  rosrun rviz rviz`

```
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[ INFO] [1596640921.840822105]: rviz version 1.13.13
[ INFO] [1596640921.840846875]: compiled against Qt version 5.9.5
[ INFO] [1596640921.840852215]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1596640921.848296983]: Forcing OpenGl version 0.
[ INFO] [1596640921.987793444]: Stereo is NOT SUPPORTED
[ INFO] [1596640921.987852694]: OpenGl version: 4.6 (GLSL 4.6).
```
I am running inside the full desktop docker image for Melodic and my GPU is a Radeon RX 570

I'm launching the container with the gui-docker sh script

So all that being said; there definitely could be something not happy around GPU processing in that setup

myFree, 2020-08-07:
I am able to run octoviz from within my docker container; so that is another sign that the GPU works from within

MrPengR, 2020-12-04:
Hello myFree, I met the same problem. Have you solved it?

myFree, 2020-12-04:
No; I have not been able to resolve it. Seems quite likely related to running with a Radeon GPU

MrPengR, 2020-12-05:
Hi. I investigated it the whole day and I found the solution. But for the deeper reason behind, I'm still confused. I would like to share it, maybe you know the answer.

**Error reason:**
The problem comes from the file "cylinder_segment.cpp" under the src folder of perception_pipeline. There is a private variable with a data type of structure pointer called "cylinder_params", which is also mentioned the tutorial. It's declared as  "AddCylinderParams* cylinder_params". In the callback function "cloudCB", this variable will be used as  "cylinder_params->direction_vec[0] = coefficients_cylinder->values[3];", which is exactly where the error comes from. It is impossible to assign a value to a structure pointer like "cylinder_params->direction_vec[0]". 

( But i don't understand why. If you like to investigate it, here is some experience of mine: 
instead of running command "roslaunch moveit_tutorials detect_and_add_cylinder_collision_object_demo.launch
", you could run "roslaunch moveit_tutorials obstacle_avoidance_demo.launch
" and in another shell run "rosrun moveit_tutorials cylinder_segment", which does exactly the same thing as written in the launch file. You could see the error message from the second shell, which says "segmentation fault (core dumped)". The error happens exactly after the code "cylinder_params->direction_vec[0] = coefficients_cylinder->values[3];". The error message "segmentation fault (core dumped)" has something to do with something like accessing data from a wrong address. This explains why using a pointer of structure raise an error here. But I think it's a bug somewhere in ROS and I don't understand why I can't do it here.)

**My solution:**
I changed the variable "cylinder_params" from a pointer of the structure to a structure. Declare it as "AddCylinderParams cylinder_params;". And in the whole file, change "cylinder_params->"  
 to "cylinder_params.". Just "call by value or call by reference" thing. Then the program works.

Best regards!

MrPengR, 2020-12-06:
Sorry I think my solution isn't right. The solution did remove the errors but did not realize the function. Because in my solution, if not  using a pointer, the value won't be changed in the callback function. And as a result, the expected cylinder won't show up in RViz because its radius and height both are not correctly assigned with a value. 

Best regards!

myFree, 2020-12-07:
Thank you for the input regardless; seems like you are really digging into finding a solution. Do you have a Radeon or NVIDIA graphics card?

MrPengR, 2020-12-07:
Mein is NVIDIA. And I installed cuda last week. Not sure if it's related.