moveit/moveit #2237: "No joint given (namespace: /arm_position_controller)" Gazebo Simulation

dhiegomaga, 2020-08-06:
### Description

I am having such a terrible experience with this software stack that I decided to post this issue here. 

If you follow the [MoveIt setup assistant tutorial](https://ros-planning.github.io/moveit_tutorials/doc/setup_assistant/setup_assistant_tutorial.html), and try to simulate the robot on gazebo using the generated gazebo.launch files, it doesn't work. 

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16
* Binary version. 

### Steps to reproduce

_(After following the MoveIt setup assistant tutorial steps... )_

- Firstly, the generated `gazebo.launch` file does not include the correct `robot_gazebo.urdf`. You need to manually copy this `.urdf`code given at the end of the set up assistant to a file. If you don't do this, you won't be able to reproduce the issue. 

- Also, you get these weird errors, I'm not sure they are related: 

```bash
[ERROR] [1596722338.944880719, 0.129000000]: GazeboRosControlPlugin missing <legacyModeNS> while using DefaultRobotHWSim, defaults to true.
This setting assumes you have an old package with an old implementation of DefaultRobotHWSim, where the robotNamespace is disregarded and absolute paths are used instead.
If you do not want to fix this issue in an old package just set <legacyModeNS> to true.
```

To fix this, you need to add to the `robot_gazebo.urdf` the following: 

```bash
<gazebo>
   <plugin name="gazebo_ros_control" filename="libgazebo_ros_control.so">
       <robotNamespace>/</robotNamespace>
       <robotSimType>gazebo_ros_control/DefaultRobotHWSim</robotSimType>
       <legacyModeNS>true</legacyModeNS>
   </plugin>
</gazebo>
```

- *(assuming you saved the generated moveit package to your `src/` folder)* If you try to execute `roslaunch yourobot_moveit_config demo_gazebo.launch`, you get error messages that are at the very beginning of the execution and might go unnoticed: 

```bash
[ERROR] [1596721225.134301059, 0.442000000]: No joint given (namespace: /arm_position_controller)
[ERROR] [1596721225.134336829, 0.442000000]: Failed to initialize the controller
[ERROR] [1596721225.134366441, 0.442000000]: Initializing controller 'arm_position_controller' failed
[ERROR] [1596721226.135438, 1.347000]: Failed to load arm_position_controller
```

- Then move your robot end-effector inside Rviz and click "Execute" to see the error. 

### Expected behaviour
Robot simulation in gazebo should move. 

### Actual behaviour
You get an error on the console:

```bash
Unable to identify any set of controllers that can actuate the specified joints: [ joint_1 joint_2 joint_3 joint_4 joint_5 joint_6 ]
[ERROR] [1596722756.322417325, 14.818000000]: Known controllers and their joints:

[ INFO] [1596722756.360914281, 14.827000000]: ABORTED: Solution found but controller failed during execution
```

simonbogh, 2021-02-09:
There are several issues here, and some related to moveit setup assistant. (If you just want a working Franka robot with moveit and Gazebo, see at the bottom where you can download it.)

### Moveit setup assistant and the tutorial
Here are the problems I found and how to fix it. Optimally, some of this should be fixed in moveit setup assistant.
1. In `demo_gazebo.launch`, you have to set the flag `load_robot_description` to `true`, unless you load the URDF model yourself, else the robot description (URDF) is not loaded. The actual loading is done in `planning_context.launch`
2. `gazebo.launch` also tries to load `robot_description` regardless of the flag above and the push it to Gazebo. It should respect the load argument. Furthermore, it does not load the URDF with `xacro` but `textfile`, hence the XACRO is not processed and `robot_description` becomes malformated which means the robot never shows up in Gazebo
3. When adding controllers in moveit setup assistant, they are generated in the file `ros_controllers.yaml`. However, you need both controllers under `controller_list:` and at the root of the file e.g. `panda_arm_controller:`. However, moveit setup assistant will not allow you to add two controllers with the same name, so you probably have to add this manually after generating the package

### Generating a working moveit_config for use with Gazebo
Here is my little checklist, for now, to make it work.
1. `demo_gazebo.launch`: set `load_robot_description` to `true`
2. `planing_context.launch`: if you use the franka_description from the PR in the next section below, you need to set the flag `use_gazebo_sim` to `true`, something like this: `<param if="${arg load_robot_description}" name="${arg robot_description}" command="xacro '${find franka_description}/robots/panda_arm_hand.urdf.xacro' use_cylinder_collision_model:=false use_gazebo_sim:=true"`. If you use the currently available official franka_description, then you need to add the Gazebo parameters you generate in the moveit setup assistant GUI
3. `ros_controllers.yaml`: add controllers under both `controller_list:` and at the root of XML-file. Also, tune the PID gains, as they are really low
4. `ros_controllers.launch`: Make sure the controllers are actually listed in the controller_spawner under args e.g. `args="panda_arm_controller panda_hand_controller"`. I have no idea why this is not filled by moveit setup assistant

### Downloading working setup (currently PRs waiting for approval)
If you just want something that works now and skip the setup assistant, use these two PRs waiting for approval, one adds the proper Gazebo tags and transmissions to `franka_description`, the other is an updated `panda_moveit_config` to support that with the right controllers.
* https://github.com/frankaemika/franka_ros/pull/126
* https://github.com/ros-planning/panda_moveit_config/pull/68

I hope these PRs will be accepted soon. I also that the bugs above will be fixed in setup assistant, because it could be a nice tool, but it only gets you 95% there, and the last 5% are super difficult to debug and get to work.

felixvd, 2021-02-11:
@simonbogh Did you have a read through [this PR](https://github.com/ros-planning/moveit_tutorials/pull/499)? An extra review would be helpful. Most of the maintainers don't seem to touch Gazebo a lot, and I don't have the cycles to take a closer look right now either.

rhaschke, 2022-10-05:
I think this has improved a lot.