moveit/moveit #2238: Crash on negative sizes in URDF

simonschmeisser, 2020-08-06:
### Description

Somewhat recently (https://github.com/ros-planning/geometric_shapes/commit/07a5254b698ecc89e511dc5b3182ecb6c7d2258e) a check was introduced whether `Shapes` have negative lengths. This crashes whoever displays that URDF (`RobotStateDisplay`)

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Version 1.0.5

### Steps to reproduce
modify some URDF to have a eg cylinder with negative length

### Expected behaviour
Display a warning in the side pane. I guess. For experienced user I guess a crash is a fine way to notify them but it also brings down our URDF editor which is used by novice people who might be frustrated then. I'll therefore add a try catch in our wrapper of `RobotStateDisplay` but was wondering if I should also open a PR

### Actual behaviour
it crashes

felixvd, 2020-08-07:
This might be why the setup assistant has recently started crashing when I try to save a new robot pose in our moveit config. Would love if you filed a PR if you already know what to fix.

simonschmeisser, 2020-08-07:
You shouldn't normally have negative sizes in URDF right? Should be easy to check your URDF/xacro @felixvd

felixvd, 2020-08-07:
I don't see why I would have any in there, but since no one else reported an issue with the setup assistant I assume it's something with our URDF which I don't have a close eye on. But it's kind of a long chain of includes and it's not been high priority enough for me to fix.

gavanderhoorn, 2020-08-07:
> since no one else reported an issue with the setup assistant

Well .. ros-industrial/universal_robot#527.

felixvd, 2020-08-07:
Well, on this repository... 

For what it's worth, I have URs in that moveit_config and it loads.

PR #2468 (2020-12-28): Catch exceptions during RobotModel loading and display them in rviz
### Description

This fixes #2238 : 
> Somewhat recently (https://github.com/ros-planning/geometric_shapes/commit/07a5254b698ecc89e511dc5b3182ecb6c7d2258e) a check was introduced whether `Shapes` have negative lengths. This crashes whoever displays that URDF (`RobotStateDisplay`)

[image]
You can see the error message from the exception and also that negative dimensions are not checked in the code path for `rviz::Robot` and Ogre actually handles them quite fine (in contrast to fcl so it's good that we do have those checks).

# [... cut]
closed 2020-12-28 by rhaschke