moveit/moveit #2241: How to silence fix / silence clang-tidy "-Woverloaded-virtual" warning in ompl interface

JeroenDM, 2020-08-08:
Context: I'm adding a modified version of the `StateValidityChecker` in the OMPL interface for constrained planning, and with so many warnings, it is difficult to see when I'm doing something wrong.

[These lines](https://github.com/ros-planning/moveit/blob/dc909df5d2cf93868a1773db869e6767e339cf7a/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/detail/state_validity_checker.h#L64) in `state_validity_checker.h` of the OMPL interface cause warnings:

>In file included from /home/jeroen/ros/moveit_ws/src/moveit/moveit_planners/ompl/ompl_interface/src/detail/state_validity_checker.cpp:37:
/home/jeroen/ros/moveit_ws/src/moveit/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/detail/state_validity_checker.h:64:8: warning: 'ompl_interface::StateValidityChecker::isValid' hides overloaded virtual function [-Woverloaded-virtual]
  bool isValid(const ompl::base::State* state, bool verbose) const;
       ^
/home/jeroen/ros/moveit_ws/devel/include/ompl-1.5/ompl/base/StateValidityChecker.h:128:26: note: hidden overloaded virtual function 'ompl::base::StateValidityChecker::isValid' declared here: different number of parameters (4 vs 2)
            virtual bool isValid(const State *state, double &dist, State *validState, bool &validStateAvailable) const

(For all of them, see [this gist](https://gist.github.com/JeroenDM/82f37c619cac54f5d51ed49f8408e049).)

This [stackoverflow](https://stackoverflow.com/questions/18515183/c-overloaded-virtual-function-warning-by-clang) post describes the problem, but I think the solution is not applicable here because we want to override some of the `isValid` methods, and reused some other from the parent class, **and** define new variants in the child class.

@tylerjw any suggestion? :)

Or should I post these types of questions on Stackoverflow?

JeroenDM, 2020-08-08:
**Note**: adding the solution from Stack Overflow does silence the warnings
```C++
class StateValidityChecker : public ompl::base::StateValidityChecker
{
public: // or protected. (not private as I need it in a derived class)
  using ompl::base::StateValidityChecker::isValid;

public:
  StateValidityChecker(const ModelBasedPlanningContext* planning_context);

  bool isValid(const ompl::base::State* state) const override
  {
    return isValid(state, verbose_);
  }

  bool isValid(const ompl::base::State* state, double& dist) const override
  {
    return isValid(state, dist, verbose_);
  }

  virtual bool isValid(const ompl::base::State* state, bool verbose) const;
  virtual bool isValid(const ompl::base::State* state, double& dist, bool verbose) const;
// the rest of the class
};
```
But think this changes the semantics of the code, and I'm not sure it is still correct in this case

rhaschke, 2020-08-10:
This looks good to me. In which sense, do you think, the semantics of the code is changed?

JeroenDM, 2020-08-10:
Maybe it was *not* our intention to make all variants of `isValid` from the base class available. But as it publicly inherits from the base class, that *is* the intention probably.

I assume this override
```C++
bool isValid(const ompl::base::State* state) const override;
bool isValid(const ompl::base::State* state, double& dist) const override;
```
 will effectively override the ` using ompl::base::StateValidityChecker::isValid;` statement for those specific versions.

v4hn, 2020-08-10:
If we actually want to keep and forward all the function variants without different names, the `using` approach is fine.

If you create a pr for this, please keep the declarations together and add a comment to the `using` line to explain why it's needed.

JeroenDM, 2020-08-10:
The `StateValidityChecker` is [passed to OMPL](https://github.com/ros-planning/moveit/blob/576c39c3d82744917123cab39dda99ba6e272a6d/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L124) and mainly used in the OMPL source code, not on the MoveIt side. So I think it we should forward all function variants, as we have no direct control over which one will be used.

I'll create a pull request tomorrow.

JeroenDM, 2020-08-11:
I have difficulty reproducing the warning in a clean workspace. I think I'll wait on the results of the Travis build for this pull request #2247 to see if it needs fixing.