moveit/moveit #2250: Path constraints not working with attached body

cpt-yoshi, 2020-08-12:
### Description
The description here will focus on the `OrientationConstraint`, but this problem should also affect `PositionConstraint`. The issue was first flagged in [MoveitTaskConstructor issue #191](https://github.com/ros-planning/moveit_task_constructor/issues/191). 

The orientation constraint given here is to make sure that the water in the attached object `cuvette` does not spill water in the workspace. However this does not work if the `link_name` is an attached object or a sub-frame.
``` c++
moveit_msgs::Constraints constraint;
constraint.name = "test";

Eigen::Quaterniond quat = Eigen::AngleAxisd(0, Eigen::Vector3d::UnitZ()) *
                          Eigen::AngleAxisd(0, Eigen::Vector3d::UnitY()) *
                          Eigen::AngleAxisd(0, Eigen::Vector3d::UnitX());

moveit_msgs::OrientationConstraint oc;
oc.header.frame_id = "optical_table"; // See below left image
oc.link_name = "cuvette";  // Object that is attached to the eef, see below right image
oc.orientation.w = quat.w();
oc.orientation.x = quat.x();
oc.orientation.y = quat.y();
oc.orientation.z = quat.z();

oc.absolute_x_axis_tolerance = 0.4; // ~23 degrees
oc.absolute_y_axis_tolerance = 0.4; // ~23 degrees
oc.absolute_z_axis_tolerance = 2 * M_PI; // Represents the gravity axis (Pointing UP)
oc.weight = 1;
constraint.orientation_constraints.push_back(oc);
```
[image][image] 

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source or Binary build? source
* If binary, which release version?
* If source, which branch? master

### Steps to reproduce
Add an object and attach it to you eef. Create an orientation constraint from point A to point B. You could use MTC to plan and visualize the results to see multiple solutions because there are some valid solutions returned. (See expected and actual behavior)

### Expected behaviour
Open the file `fixed_orientation_constraint.mp4` from this [link](https://drive.google.com/drive/folders/1VG8RhXeWHGmfgLjtKWfClQ4K2aDuKhEe?usp=sharing). You will see that the orientation constraint is respected for multiple cases and that there are no false negative/positive. Compared with the actual behavior below.

### Actual behaviour
Open the file `default_orientation_constraint.mp4` from this [link](https://drive.google.com/drive/folders/1VG8RhXeWHGmfgLjtKWfClQ4K2aDuKhEe?usp=sharing). Some orientation constraints are not respected.

### Backtrace or Console output
I have added backtraces for different path that triggers `OrientationConstraint::configure` and `OrientationConstraint::decide`:
[Bactrace#1](https://gist.github.com/cpt-yoshi/38d608e4a0e674982dacd17f8c49d2ed)
[Bactrace#2](https://gist.github.com/cpt-yoshi/df647f3240e1a4dbde02bbc3885a3072)
[Bactrace#3](https://gist.github.com/cpt-yoshi/469100ba9aa2946639e6b2d5166c2c90)

### Analysis
The [generatePlan](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/planning_pipeline/src/planning_pipeline.cpp#L210-L213) will trigger the [isStateValid](https://github.com/ros-planning/moveit/blob/master/moveit_core/planning_scene/src/planning_scene.cpp#L2100-L2101) which will trigger the  `resolveConstraintFrames`. If successfull, it will then trigger the [isPathValid](https://github.com/ros-planning/moveit/blob/master/moveit_core/planning_scene/src/planning_scene.cpp#L2162-L2165) which will **NOT** trigger the `resolveConstraintFrames`. So when the Orientation::configure tries to get the link model, it fails to retrieve the link because it is an attached object.

I think there is another issue at hand. There is clearly something wrong with the `isStateValid` and/or the `IKConstraintSampler::validate`. Because when something fails there, the `isPathValid` is not triggered. And when I force these 2 validation to always pass, there are a lot of new good solutions found! See videos above.

When all the above is fixed, I think there is another problem, because the `isPathValid` through [OrientationConstraint::decide](https://github.com/ros-planning/moveit/blob/master/moveit_core/kinematic_constraints/src/kinematic_constraint.cpp#L619) gives bad solutions. Here is the fix the problem:

``` c++
// When the ResolvedConstraintFrames changes the oc.orientation...

// Actual
// desired_rotation_matrix_ = world_R_desired * object_R_link
// desired_rotation_matrix_inv_ = link_R_object * desired_R_world
// diff rotation =  link_R_object * desired_R_world * world_R_link
Eigen::Isometry3d diff(desired_rotation_matrix_inv_ * state.getGlobalLinkTransform(link_model_).linear());

// Fix
// desired_rotation_matrix_ = world_R_desired
// desired_rotation_matrix_inv_ = desired_R_world
// diff rotation =  desired_R_world * world_R_link * link_R_object
Eigen::Isometry3d diff(desired_rotation_matrix_inv() * state.getGlobalLinkTransform(link_model_).linear() * link_to_object_rotation_matrix_);
```
I think we need to keep an extra rotation in the OrientationConstraintMsgs for the original desired_rotation...

I have this [branch](https://github.com/cpt-yoshi/moveit/tree/orientation_constraint) that fixes this issue. Note that this is a **LAZY   UGLY HACK**, it only fixes the issue for the OrientationConstraint when the link_name is fixed. Also, read the Readme for this hack to work.

### Discussion
As discussed with @v4hn, we should probably add the ResolvedConstraints before the planning pipeline. This would reduce calls to ResolvedConstraints that is triggered **2** times at the moment (**3** times  if you fix the isPathValid). Also there would be no need to add the ResolvedConstraints adapter for each planning pipeline.

When the `isStateValid` and the `IKConstraintSampler::validate` are forced to **PASS**, I don't understand why it fixes some issues. I think that these validations, for path_constraints, is not necessary because it is rechecked in `isPlanValid` for the trajectory. This clearly speedup quite a bit the process (See videos and compare the task time). This is based on a **limited** knowledge of this library though...

gautz, 2021-11-16:
Now that [subframes are being handled  correctly by MTC](https://github.com/ros-planning/moveit_task_constructor/pull/304), what would be the next step to get the path constraints correctly transformed? E.g. for setting CIRC constraints with Pilz Industrial Motion Planner or Cartesian constraints with OMPL constrained planning. 
@captain-yoshi @v4hn ?

gautz, 2021-11-16:
Is #2890 a workaround to solve this issue for Pilz?

captain-yoshi, 2021-11-16:
@gautz I'm planning to benchmark path constraints (OMPL, Pilz) in 2 weeks and hopefully fix this issue (if the issue still exists) ! Reading the workaround you mentioned, it seems that Pilz dose not pass through the `KinematicConstraint` class so it should work. Will report back.

gautz, 2021-11-16:
An example of the remaining issue (partly solved by [e30c6fa](https://github.com/ros-planning/moveit/commit/e30c6fa853dbc699c0cc84e420626000eefe3212) and #2890)

When setting setIKframe to an attached object subframe, the interim point is not correctly transformed
```
pos_constraint.constraint_region.primitive_poses.push_back(interim_frame_.pose);
path_constraints.name = pilz_circ_third_point_type_;
```
https://user-images.githubusercontent.com/64833674/142002683-a90bf9ef-076a-4c81-839e-a750961bacff.mp4

When I do not use setIKframe, the CIRC motion is OK:
https://user-images.githubusercontent.com/64833674/142002653-aecb6d74-c212-4f5a-92f0-75e41b5a75a9.mp4

captain-yoshi, 2021-11-25:
@gautz Can you validate if PR #2975 fixes Cartesian constraints with OMPL constrained planning ?

gautz, 2021-11-29:
Still getting the same error with Pilz.
With OMPL something is wrong with the constrained state space as the resulting trajectory has joint space jumps :
https://user-images.githubusercontent.com/64833674/143878319-2a4a7cc1-9a55-48a5-8a62-2d75890aa6c1.mp4

Maybe @JeroenDM or @zkingston would like to have a look?

captain-yoshi, 2021-11-29:
This joint space jump is a know issue. By default, orientation path constraint are sampled in [cartesian space](https://ros-planning.github.io/moveit_tutorials/doc/ompl_interface/ompl_interface_tutorial.html#enforce-planning-in-joint-space). You can enforce this planning in joint space by setting `enforce_joint_model_state_space: true` in the ompl_planning.yaml, but this might increase planning time.

Jeroen added a [new approch for planning with path constraints](https://moveit.ros.org/moveit/2020/09/10/ompl-constrained-planning-gsoc.html). I think the orientation path constraint has been merged ? If so you can use it with `enforce_constrained_state_space: true`.

> Now that subframes are being handled correctly by MTC, what would be the next step to get the path constraints correctly transformed? E.g. ... Cartesian constraints with OMPL constrained planning.

@gautz Aside from the joint space jump depending on the ompl planning config you have, is cartesian path constraint correctly transformed with or without #2975 ?

gautz, 2021-11-29:
I am actually setting position constraints. Orientation constraints are not yet available. This would actually be my next step.

I just double checked:
1. without #2975 path constraints are not correctly transformed but there is no joint space jump
2. with #2975 path constraints are correctly transformed
BUT:
3. with #2975 there is a big regression: joint space jump is occurring most of the time

gautz, 2021-11-29:
Am I supposed to modify something in my MoveIt setup? Like remove `default_planner_request_adapters/ResolveConstraintFrames` from `ompl_planning_pipeline.launch.xml`

You can test your own PR by using/extending https://github.com/ros-planning/moveit_tutorials/pull/651 (comment out the stuff related to orientation constraints eventually)
I can also help out there if you need.

captain-yoshi, 2021-12-01:
Sorry ... hours became days because of segfault !

I reproduce your first example (the box-constraint) but with the `PlanningPipeline` API. Without my PR and with frames that are just robot links, there is a lot of joint flips (1 out of 2 benchmarks). Was your mistake in some of the configuration ? Using latest MoveIt from master.

ompl_planning.yaml
```yaml
panda_arm:
  default_planner_config: RRTConnect
  planner_configs:
     ...
  enforce_constrained_state_space: true
  projection_evaluator: joints(panda_joint1,panda_joint2)
  longest_valid_segment_fraction: 0.005

```

gautz, 2021-12-02:
The PR is not merged (and will probably never be as it is already in MoveIt2) 
You need to checkout (and rebase?) https://github.com/ros-planning/moveit/pull/2273

closed 2024-10-31 by rhaschke