moveit/moveit #2253: On current upstream branch `noetic-devel`

v4hn, 2020-08-13:
It looks like we split `noetic-devel` from `master` via a3994effdc7aa46e66f6b95a142c17948c98fbcf end of June this year.
I believe we actually discussed that we will release the first noetic distribution from the `master` branch though. Is this not feasible for technical reasons?

Should we consider the `noetic-devel` branch as a test bed for noetic compatibility?
If we treat it as a fully featured distribution branch, we will need to review backports for it from `master` which we did not do yet and `master` already includes quite a few fixes and structure changes by now. I would like to avoid this overhead for the near future though.

@ros-planning/moveit-maintainers

rhaschke, 2020-08-13:
`noetic-devel` currently only exists to provide Dockerfiles for Noetic. There is a corresponding PR (#2178), explaining the technical issues you are asking for and the proposed procedure.
From time to time I rebase `noetic-devel` onto `master` because we didn't make progress towards a release...
However, to be honest, this is mainly due to external packages missing (see list in #2178).

closed 2020-08-13 by rhaschke

v4hn, 2020-08-13:
Thanks for the reference. Could you possibly rename the branch to `noetic-devel-staged` or something similar to avoid users accidentally using this branch?

rhaschke, 2020-08-13:
The name is already configured on dockerhub to automatically build the Noetic images. So, I would prefer to keep that name.
Where do you see a risk in users using this branch?

v4hn, 2020-08-13:
There was [a similar approach in the universal_robot repository a while back](https://github.com/ros-industrial/universal_robot/issues/444) and some people (including me) accidentally used the wrong branch, simply because it had the branch name of the current ROS distribution. That could be avoided by a different branch name.

The dockerhub configuration is a good reason not to change it though. What about a prominent text in the README.md to state that this branch is not recommended for users?