moveit/moveit #2254: moveit servo velocities not getting set to zero when no commands are incoming

rfn123, 2020-08-13:
### Description

I have been trying to use the moveit_servo package with a UR3. I am using the `/joint_group_vel_controller` to control the robot with the servo package. I noticed that when stop publishing commands on `servo_server/delta_twist_cmds` , the velocities published to the controller topic are not getting set to zero.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source or Binary build?
* Copied the moveit_servo package from the `melodic-devel` branch and built it

### Steps to reproduce

I start the node with this launch file, with `sim:=false` :
```xml
<launch>

  <arg name="sim" default="true" />

  <node name="servo_server" pkg="moveit_servo" type="servo_server" output="screen" >
    <param name="parameter_ns" type="string" value="servo_server" />
    <rosparam command="load" file="$(find moveit_servo)/config/ur_simulated_config.yaml" if="$(arg sim)"/>
    <rosparam command="load" file="$(find moveit_servo)/config/ur_real_config.yaml" unless="$(arg sim)"/>
  </node>

</launch>
```

Here is the configuration file I used, I hope everything is correct there:
```yaml
###############################################
# Modify all parameters related to servoing here
###############################################
use_gazebo: false # Whether the robot is started in a Gazebo simulation environment

## Properties of incoming commands
robot_link_command_frame:  ee_link  # commands must be given in the frame of a robot link. Usually either the base or end effector
command_in_type: "speed_units" # "unitless"> in the range [-1:1], as if from joystick. "speed_units"> cmds are in m/s and rad/s
scale:
  # Scale parameters are only used if command_in_type=="unitless"
  linear:  0.0006  # Max linear velocity. Meters per publish_period. Unit is [m/s]. Only used for Cartesian commands.
  rotational:  0.003 # Max angular velocity. Rads per publish_period. Unit is [rad/s]. Only used for Cartesian commands.
  # Max joint angular/linear velocity. Rads or Meters per publish period. Only used for joint commands on joint_command_in_topic.
  joint: 0.001
low_pass_filter_coeff: 2.  # Larger --> trust the filtered data more, trust the measurements less.

## Properties of outgoing commands
publish_period: 0.017  # 1/Nominal publish rate [seconds]
publish_delay: 0.005  # Delay between calculation and execution start of command

# What type of topic does your robot driver expect?
# Currently supported are std_msgs/Float64MultiArray (for ros_control JointGroupVelocityController or JointGroupPositionController)
# or trajectory_msgs/JointTrajectory (for Universal Robots and other non-ros_control robots)
command_out_type: std_msgs/Float64MultiArray

# What to publish? Can save some bandwidth as most robots only require positions or velocities
publish_joint_positions: false
publish_joint_velocities: true
publish_joint_accelerations: false

## MoveIt properties
move_group_name:  manipulator  # Often 'manipulator' or 'arm'
planning_frame: base_link  # The MoveIt planning frame. Often 'base_link' or 'world'

## Stopping behaviour
incoming_command_timeout:  0.5  # Stop servoing if X seconds elapse without a new command
# If 0, republish commands forever even if the robot is stationary. Otherwise, specify num. to publish.
# Important because ROS may drop some messages and we need the robot to halt reliably.
num_outgoing_halt_msgs_to_publish: 0

## Configure handling of singularities and joint limits
lower_singularity_threshold:  17  # Start decelerating when the condition number hits this (close to singularity)
hard_stop_singularity_threshold: 30 # Stop when the condition number hits this
joint_limit_margin: 0.15 # added as a buffer to joint limits [radians]. If moving quickly, make this larger.

## Topic names
cartesian_command_in_topic: servo_server/delta_twist_cmds  # Topic for incoming Cartesian twist commands
joint_command_in_topic: servo_server/delta_joint_cmds # Topic for incoming joint angle commands
joint_topic:  joint_states
status_topic: servo_server/status # Publish status to this topic
command_out_topic: /joint_group_vel_controller/command # Publish outgoing commands here

## Collision checking for the entire robot body
check_collisions: true # Check collisions?
collision_check_rate: 40 # [Hz] Collision-checking can easily bog down a CPU if done too often.
# Two collision check algorithms are available:
# "threshold_distance" begins slowing down when nearer than a specified distance. Good if you want to tune collision thresholds manually.
# "stop_distance" stops if a collision is nearer than the worst-case stopping distance and the distance is decreasing. Requires joint acceleration limits
collision_check_type: stop_distance
# Parameters for "threshold_distance"-type collision checking
self_collision_proximity_threshold: 0.02 # Start decelerating when a self-collision is this far [m]
scene_collision_proximity_threshold: 0.05 # Start decelerating when a scene collision is this far [m]
# Parameters for "stop_distance"-type collision checking
collision_distance_safety_factor: 1000 # Must be >= 1. A large safety factor is recommended to account for latency
min_allowable_collision_distance: 0.02 # Stop if a collision is closer than this [m]
```

### Expected behaviour
When no commands are incoming the velocities are being set to zero --> Robot stops.

### Actual behaviour
Velocities remain the same as before --> Robot continues moving

### Backtrace or Console output
The console output of the servo_server node. I noticed that when stopping publishing Twist messages no stale warning is printed, but rather they are printed when I _start_ publishing which I find a bit strange. 
```
[ INFO] [1597322665.834443871]: Listening to '/attached_collision_object' for attached collision objects
[ INFO] [1597322665.847292816]: Waiting for first joint msg.
[ INFO] [1597322666.009899317]: Received first joint msg.
[ WARN] [1597322666.415038447]: last_duration: 0.0370889 (0.025)
[ WARN] [1597322679.669836648]: Stale command. Try a larger 'incoming_command_timeout' parameter?
[ WARN] [1597322694.374796934]: Stale command. Try a larger 'incoming_command_timeout' parameter?
[ WARN] [1597322696.560582991]: last_duration: 0.0327494 (0.025)
```

rhaschke, 2020-08-13:
Fixed via #2255.

closed 2020-08-13 by rhaschke

Altaibaatar, 2024-04-25:
It's happening on ROS2 Humble.

rhaschke, 2024-04-26:
> It's happening on ROS2 Humble.

@Altaibaatar, this is the MoveIt for ROS1 repo. File an issue at https://github.com/ros-planning/moveit2 if you having problems on ROS2.