moveit/moveit #2259: Moveit Pick - Grapper closes into collision object and fails to grap

Fatalon, 2020-08-14:
### Description
* same as https://answers.ros.org/question/359566/moveit-pick-grapper-closes-into-collision-object-and-fails-to-grap/

Hello. I am using a Franka Emika robot with a gripper mounted as eef and control his movements in the simulation environment gazebo via the moveit package. I rebuilt the entire scene in rviz (including all collision objects) to estimate valid actions in moveit. But when I try to grap a collision object (in my case it is a hanoi tower token, which is put onto a pipe) - the robot moves into the appropriate pose, opens the gripper - and closes it, ignoring the properties of the token - resulting in an abort and pushing the token in gazebo away. I followed the tutorial instructions (http://docs.ros.org/kinetic/api/movei...) and also tried to adjust the max contact force or directly change the properties of the closedGrapper, but nothing really worked out. 

### Your environment
* ROS Distro: [Melodic]
* Ubuntu 18.04
* ros-melodic-moveit-ros-move-group/bionic,now 1.0.5-1bionic.20200803.145038 amd64 [installed]

### Steps to reproduce
* i have my source-code available in my gitlab repo at https://gitlab.com/Fatalon/pandarobottowerofhanoi

* very similar code to pick-and-place tutorial
* adjusted values to my scene, including all new collision objects
* calling grap on the token fails with "Controller panda_hand_controller failed with error GOAL_TOLERANCE_VIOLATED

### Expected behaviour
* the grapper should stop when it has contact with the object and therefore pick it up

### Actual behaviour
* the grapper does not stop and completely closes the gripper again 

[image] - rviz
[image] - gazebo

felixvd, 2020-08-17:
Thanks for linking to your ROS Answers post, but I cannot open the link. I am guessing your thread was [moved to Gazebosim](https://answers.gazebosim.org/question/25688/moveit-pick-grapper-closes-into-collision-object-and-fails-to-grap/) because it is most likely a Gazebo problem and not a bug in MoveIt.

By default, MoveIt does not calculate how far the gripper should close to grasp a given object. The pick pipeline assumes that your gripper knows not to violate the laws of physics when sending a "close" command. The behavior you describe is not surprising if you are commanding the finger pads to move to a fully closed position. The finger joints in Gazebo would follow those commands and force the object out of the way.

To stop closing the gripper while grasping an object (which would happen automatically in the real world when the object is in the way), you need to set up your gripper controller accordingly in Gazebo. This is out of scope here, but I would start by looking into the EffortJointInterface and friction parameters. Related links: [1](https://answers.ros.org/question/291228/simple-box-grasping-fails/), [2](https://answers.ros.org/question/290398/help-with-procedure-for-simple-pick-and-place-task-using-moveit/), [3](https://answers.ros.org/question/353487/gripper-controllers-for-franka-panda-in-gazebo-simulation/), [4](https://dof.robotiq.com/discussion/1691/using-2f-85-gripper-with-sawyer-in-gazebo)

I'll close this since I don't think this is a bug, just a difficult thing to implement. Please update the post if you disagree, and do update your gazebosim thread with your solution so the community can benefit.

closed 2020-08-17 by felixvd

mfheinemann, 2022-08-18:
Hi, I am having the same issue: I am trying to pick some boxes of variable sizes with my OpenManipulator-P. I am using the GripperActionController and everything works fine until the gripper gets blocked by the box. I am commanding the gripper to fully close (angle 1.1 rad) since I do not want to calculate exactly the angle at which the gripper perfectly holds each box. Since the gripper commander is then sending "ABORTED: Controller failed during execution" Moveit assumes grasping failed. In my opinion Moveit should register the blocked gripper and count this as a success.

Maybe there is a solution that I overlooked but I searched a lot online and tried a lot of things. Maybe someone here can help.

Thanks in advance!

Setup: ROS Noetic on Ubuntu 20.04; OpenManipulator-P with gripper

rhaschke, 2022-08-18:
@mfheinemann, adapt the goal tolerance of your action controller! MoveIt just reacts to the results of the action controllers. If the controller considers the execution a failure, MoveIt of course does so as well.