moveit/moveit #2261: Plan problems during simplifySolutions

akjay, 2020-08-18:
During the plan process,we use such as RRT or RRTConnect algorithm to find a path,after random search,we will do a job named "simplifySolution", it will call function in ompl::geometric::PathSimplifier class to do path simplify work.
In this function, there are many ways to simplify a solution,"shortcutPath" ,"findBetterGoal", "smoothBSpline", "checkAndRepair", etc..Each step means to reduce vertices in the path,and it really makes the path better to be executed.
But there seems a problem that,when we try to reduce a vertex or replace one with new state,we will check whether the state is valid or not,like `checkMotion(state s0, state s1)`,this function will check states interpolate among the s0 and s1,the states count is calculate by function `validSegmentCount`,but in many situation,after the simplifySolution work, the path is been reduced only 3 or 4 states left, however the minimum_waypoint_count is 10,so we need to interpolate again to meet the minimum_waypoint_count requirement.Sometimes,the total `validSegmentCount` between the simplified path states is less than 10,then we need to do a fine interpolate to meet the requirement,but this will generate some states that not checked before,and this may cause a collision when final path check after interpolation,this will cause the path invalid and can not be executed.
So maybe the initial path can work,but after simplify work,the path turned to be invalid.How can we improve simplify work to prevent this happen?Do more check during simplify steps?

v4hn, 2020-08-18:
If you experience problems with this additional interpolation, you can set the minimum_waypoint_count to 2.
But that of course will not remove collisions if they are real.
The same way you can also disable simplify_solutions altogether.

But the problem that simplified solutions might suddenly collide is a very general one and I'll forward it to @mamoll . :-)

akjay, 2020-08-18:
I think it not a good idea that set the minimum_waypoint_count to 2,because the collisions is really exist,set it to 2 is just reduce the error report,but if you really move the robot follow this trajectory,the collision will happen.
Also, if disable simplify solutions,the path is not friendly for robot to move,too many vertices and not smooth, sometimes the path will  shuttle between two points many times, the random path seems a little silly.
I have a temporary solution is that check as more states as we can during the simplify process,no matter how many counts `validSegmentCount` will return, once when we need to `checkMotion` between two states, set the count to 10(equal to minimum_waypoint_count),make sure we check enough states, in this case, if the `checkMotion` return true, we do the simplify work.

v4hn, 2020-08-18:
> I think it not a good idea that set the minimum_waypoint_count to 2,because the collisions is really exist,set it to 2 is just reduce the error report,but if you really move the robot follow this trajectory,the collision will happen.

Yes, I agree.

> Also, if disable simplify solutions,the path is not friendly for robot to move,too many vertices and not smooth, sometimes the path will  shuttle between two points many times, the random path seems a little silly.

This of course depends on the planner, but yes, path simplification exists for a reason.

> I have a temporary solution

This boils down to your `longest_valid_segment_fraction` parameters being too high in general then.
If these value(s) are too high for your setup, even your planned solution trajectory can miss collisions and simplification can create new (or even remove existing) collisions in the solution path.

akjay, 2020-08-18:
> This boils down to your `longest_valid_segment_fraction` parameters being too high in general then. If these value(s) are too high for your setup, even your planned solution trajectory can miss collisions and simplification can create new (or even remove existing) collisions in the solution path.

Yes,I'm not very clear with the planner parameters, so I used the default values to plan, just change the `range` value for the planner.As you said, if the parameters too high, the trajectory will miss collisions. I will change the `longest_valid_segment_fraction` parameters to test with my robot.Thank you.

closed 2020-08-19 by akjay