moveit/moveit #2262: Moveit collision object pass through meshes

YannickRiou, 2020-08-18:
### Description

While using Moveit Task Constructor framework (based on Moveit), it appears that Moveit doesn't care about collision between a primitive collision object and a mesh collision object, at it find a "successful" plan solution. But when coming to execution, Moveit stops it as it find the primitive object is crashing into mesh collision object.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Moveit branch : Master

Issue also open for MTC as it may be related (dont know if it's moveit or MTC the problem for now...) :  https://github.com/ros-planning/moveit_task_constructor/issues/180

### Steps to reproduce

Load [this box as collision object](https://catdrop.drycat.fr/r/PQvBMYDm#LlTgUfNTZHzzrc2+CDQpe376j5hXl4jbj1shmBQRRDE=), and try to place a cube primitive object into it.

### Expected behaviour
Planner shouldn't find a solution (if too complicated) or should avoid the mesh box.

### Actual behaviour
No robot parts (gripper fingers) seems to touch the mesh box but the primitive collision object does (and a lot). 

[image]
[image]

### Backtrace or Console output

Here is the [console output](https://gist.github.com/YannickRiou/9ba9afc05bc3fe85767609ae424618b5) showing that TRRT planner finally find a solution but when going for execution, it just fails because it founds a collision between the primitive cube and the mesh box (as seen on the two gif above)

gapbridger, 2020-08-18:
hi, we are having exactly the same problem, and we are trying to find a solution but haven't figured out a way, if you have any suggestions would you please also let me know :)

v4hn, 2020-08-18:
Hi there,

please provide a full minimal example (a ROS package) for a maintainer to compile and look at.
[In the other issue on ignored attached collisions someone provided a great minimal example](https://github.com/ros-planning/moveit/issues/1217#issuecomment-624397916), but sadly I could not reproduce the issue anymore after resolving several minor problems.

YannickRiou, 2020-08-20:
I created two examples : one using MTC and one using only moveit with a simple task here : https://github.com/YannickRiou/moveit_collision_pb/tree/moveit_only 

The results are quite the same as #1217 on the moveit only test, when reviewing the planned trajectory, the object isn't moving along with the gripper.... 

The solution found might ok (without collision) sometimes as the planner is RRTConnect, but it leads most of the times to solution with collisions.

YannickRiou, 2020-08-20:
I created two examples : one using MTC and one using only moveit with a simple task here : https://github.com/YannickRiou/moveit_collision_pb/tree/moveit_only 

The results are quite the same as #1217 on the moveit only test, when reviewing the planned trajectory, the object isn't moving along with the gripper.... 

The solution found might be ok (without collision) sometimes as the planner is RRTConnect, but it leads most of the times to solution with collisions.

When using Rviz to plan with the object attached, we can see that the planned path contains the attached collision object, and it is taken into account for planning : 
https://catdrop.drycat.fr/r/STrTI-jj#Md3gBNlzDmEcZ05OJbl0FTRyNztSLd4Mv2yOqdM6HN4=

YannickRiou, 2020-08-26:
After some more testing, it seems the problem comes from I was using master branch of moveit (v1.1.0). If instead I use 1.0.6, the solutions seems to be without collisions...

rhaschke, 2020-08-26:
Of course, the master branch shouldn't have collisions either. Please fix your example to allow debugging this.

YannickRiou, 2020-08-31:
Made some changes on the test. I exported the scene that I use on the real PR2 and that lead to solution with collision, so that I can make test on simulation. Here is the new example : https://github.com/YannickRiou/moveit_collision_pb/tree/pr2_test_scene 
Everything is explained in the readme and everything should be running without problems (if you have any please don't hesitate). 
This example is with 1.0.6 version of moveit so my previous statement about moveit master was wrong. 

A video with the full execution is available in the readme.

rhaschke, 2020-09-08:
I can confirm this issue in your example. However, I never observed this issue in other setups. Also, the issue seems to occur only in the placing phase, but not in the grasping phase, which is weird. I suspect your mesh model (having e.g. wrong normals).

To narrow the problem, I ask you to further simplify your example, just realizing a motion from A to B with a simple MTC task (`FixedState - Connect - FixedState`) or MoveIt directly. Also, try to simplify and/or repair the collision mesh (check/correct normals). Finally, you should load the initial scene programmatically. The current procedure is too cumbersome for efficient debugging.

YannickRiou, 2020-09-08:
Ok I updated the pr2_test_scene branch and simplified the setup, now the planning scene load automatically (added node in the launch file). I used only MoveTo and MoveRelative stages that don't generate multiple solution, so the execution should be the same every time. We checked earlier the mesh and everything seemed fine.

gapbridger, 2020-09-09:
hi just to give my self's input on this issue. we used RobotState::attachBody to attach the object rather than PlanningSceneInterface::applyAttachedCollisionObject and our collision problem is solved. not sure why...

v4hn, 2020-09-09:
@gapbridger I'm not sure how to interpret your statement. `RS::attachBody` operates on a local C++ class, but `PSI::applyAttachedCollisionObject` adds the object to the planning scene/robot state in the `move_group` node.

gapbridger, 2020-09-09:
@v4hn ok, so if you do `RobotState robot_state = move_group->getCurrentState();` , you can have a RobotState instance that is related to this particular move_group node right? then if you do attachBody, the attached body will change this particular RobotState instance which will (i assume) reflect the changes into this particular move_group node (we can see the attached body in RViz) since I'm not sure how the move_group node and planning_scene_interface communicate with each other, I am assuming the changes to move_group will also be captured by planning_scene_interface's back end. 

so in the end, although using PlanningSceneInterface can apply attached body to planning scene, using RobotState can also attach body to planning scene, and at least for our case, it helped to resolve the following motion planning algorithm ignores the attached body issue. 

additional info: we used two separate move_group nodes in two separate nodes where one node execute motion plan and the other node do actual planning and they communicate by action server & client. in this weird architecture, we have to use robot_state.attachBody to force the motion planner to acknowledge the attached body, we haven't been able to find out other approaches that can achieve this purpose in our architecture.

v4hn, 2020-09-09:
> if you do RobotState robot_state = move_group->getCurrentState(); , you can have a RobotState instance that is related to this particular move_group node right?

It's a local *copy* and can in no way affect the PlanningScene in the respective `move_group` node.

> if you do attachBody, the attached body will change this particular RobotState instance which will (i assume) reflect the changes into this particular move_group node

No, it will not propagate back to the move_group by itself.

> we can see the attached body in RViz

Definitely not because of the method calls you describe in your post. If you can see it in RViz, you either publish the local `RobotState` yourself or you also call some ROS interface in the resp. move_group node.

you are definitely confused about these different concepts and methods in your explanations below.

> additional info: we used two separate move_group nodes in two separate nodes where one node execute motion plan and the other node do actual planning and they communicate by action server & client

Sure, you can do that. I guess you use the namespace argument in the PlanningSceneInterface to indicate which scene you want to modify with the resp. psi.

gapbridger, 2020-09-10:
following is the code snippets where we happened to solve the problem in our project. 
state is the name of RobotState instance, and it is instantiated by move_group->getCurrentState()
not sure if by this way, somehow the change to RobotState instance can be propagated to move_group instance which in turn reflects the attach of rigid body in RViz. 

I am not really familiar with how moveit actually works (we haven't been able to go through the code base in detail), so if you can give some insight on this, it will be very much appreciated. 
`
std::vector<std::string> touch_links = {plan_frame_};
state.attachBody(names_list_[label], {shapes_list_[label]}, {relative_pose_list_[label]}, touch_links, plan_frame_);
std::vector<double> curr_plan_start_joints = previous_plan_.trajectory_.joint_trajectory.points.back().positions;
state.setJointGroupPositions(group_name_, curr_plan_start_joints);
move_group_->setStartState(state);
move_group_->setJointValueTarget(next_target_joints);
success = bool(move_group_->plan(next_target_plan));
`

v4hn, 2020-09-10:
@gapbridger  Last comment on your issue here, this is independent from @YannickRiou's issue which seems to be more fundamental.

> move_group_->setStartState(state);

This line makes `move_group` use the start state you provide (and its attached object) instead of the robot state it maintains on its own when `move_group_->plan` triggers planning.
Be aware that this uses the attached object **only** for this one request, it will overwrite any other attached bodies in the `move_group`, and you induce some additional latency by
- asking ROS/move_group for the current state
- setting it as the start state in move_groups action request
- sending it to move_group when triggering planning

The better information flow is usually to configure `move_group`'s internal scene via the `PSI::apply*` methods and invoke planning **without** a concrete start state. This way the `move_group` will fill in the start state based on what it currently sees on the joint_states topic.

gapbridger, 2020-09-11:
@v4hn how is my issue not fundamental? if PSI::apply* can work why don't we use it? we've tried it. the problem is just that this approach doesn't work and RobotState::attachBody works. 
you don't need to reply to my comments if you don't want to, i am here just to raise questions and i am replying your response simply because i think you are one of the maintainers.

v4hn, 2020-09-11:
> how is my issue not fundamental? if PSI::apply* can work why don't we use it? we've tried it. the problem is just that this approach doesn't work and RobotState::attachBody works.

I'm sorry. Apparently I came across as very rude and you misunderstood what I meant.
Reports on broken functionality are of course always important.

Please file a new issue for your problem giving some more details on your use of `PSI::apply*` in your use-case.
The other issue discussed in the thread is most likely unrelated (by "fundamental" I meant involving lower-level methods in MoveIt).
You already tracked your problem down to a concrete API not working for your use-case and this should be easier to resolve given sufficient context.

YannickRiou, 2020-09-21:
@rhaschke did you managed to run the updated pr2_test_scene branch setup ? Can you reproduce the same error ? 
On my side, still not sure that the problem isn't about how objects are attached by Moveit Task Constructor (because if we do everything manually in Rviz (move to start, attach object, publish scene, move to end), everything is fine.

rhaschke, 2020-09-21:
Sorry, I did have time yet to look into this again.

closed 2020-11-30 by YannickRiou