moveit/moveit #2263: Joints panel units in degree

v4hn, 2020-08-18:
The relatively-new joints tab to manipulate the goal state in RViz shows the joint readings in degree instead of rad.

[image]

I believe the rest of MoveIt's API uses rad exclusively to avoid confusion and unit conversions.
Shouldn't we use radian here then? Alternatively how about we add a `°` unit sign after each number.

rhaschke, 2020-08-18:
I think a user GUI should use degrees instead of radians. But, I'm fine to add a ° sign.

v4hn, 2020-08-18:
> I think a user GUI should use degrees instead of radians.

Personally, I strongly disagree because you can't use the values from there anywhere else in ROS.
I am not a user of course. Maybe @felixvd or some other power users have an opinion?

gavanderhoorn, 2020-08-18:
> you can't use the values from there anywhere else in ROS.

Actually, you can: see ros/xacro#251 fi and the linked `rosparam` docs.

v4hn, 2020-08-18:
> Actually, you can: see ros/xacro#251 fi and the linked rosparam docs.

That's a nice piece of trivia that's good to know if I ever find someone who decided to use this functionality (and somebody fixes it in xacro and possibly writes wrappers for all other places angles might appear in the ROS ecosystems).

Nonetheless, https://www.ros.org/reps/rep-0103.html#derived-units declares ROS uses SI units and `rad` as the commonly used angular unit. Whether or not rosparam has support for unit conversion does not resolve the question what we should expose in the GUI.

gavanderhoorn, 2020-08-18:
> That's a nice piece of trivia that's good to know if I ever find someone who decided to use this functionality (and somebody fixes it in xacro and possibly writes wrappers for all other places angles might appear in the ROS ecosystems).

There's no need to be so cynical about everything Michael.

It's been supported for over 12 years now, and is actually used.

v4hn, 2020-08-18:
> There's no need to be so cynical about everything Michael.

Sorry, I definitely come across as more rude than I mean to. You do know I think you're doing a great job. :smiley_cat:

> And that's actually not true.

Yes, and thanks for pointing to these mechanisms, I've learned something. Still, I would actually be more interested in your personal opinion on this issue as you are also one of the people most in touch with the entire community.

gavanderhoorn, 2020-08-18:
If you want to be nice, offer both options. But that's additional work.

At the very least, I'd suggest to just add the degree symbol. It's a very minor change, but will make it much clearer which units are being used.

As to radians being used everywhere: true, but for instance TF2 also allows users to request display in degrees (fi in `tf2_tools/echo.y`). It's just one of those things people have a more intuitive understanding of.

PR #2264 (2020-08-18): MP display: add units to joints frame
I still strictly prefer radian here, but this at least states units explicitly.

[image]

Fixes #2263 .
closed 2020-08-18 by rhaschke