moveit/moveit #2268: FixStartStateCollision can cause sudden jumps

felixvd, 2020-08-20:
### Description

Ever since I first started using MoveIt, the `FixStartStateCollision` adapter has confused me. Every time I notice it, it makes me think "Thank god that didn't hit anything".

On a UR5e, the "jiggle" it introduces can result in very fast motions of up to 10 cm. With `iiwa_stack` running on a KUKA iiwa, the jumps can go even further because of the lazy tracking. Anecdotally, I observed up to 40 cm offset from the trajectory and many surprised faces.

One can argue that if the environment is set up correctly, this won't be an issue. But there are plenty of reasons why the real world is not perfectly aligned to the planning scene, and the robot's reaction to that sort of mismatch should be to stop, not to attempt a potentially dangerous repair, so I don't think this behavior should be the default. I would suggest to:

**a)** Remove the FixStartStateCollision completely, or  
**b)** Reduce the speed of the trajectory at the start to ensure a low end effector speed, or  
**c)** Evaluate the cartesian distance of the end effector instead of the "jiggle_fraction"

One could also rewrite the adapter, but I doubt anyone has cycles for that. Would be interested in opinions.

### Steps to reproduce
Spawn an object in the scene, move it so it very slightly touches the robot, then move the robot.

### Expected behaviour
The robot should either stop or move away slowly.

### Actual behaviour
The evasive movement can be both large and very fast.

v4hn, 2020-08-20:
> But there are plenty of reasons why the real world is not perfectly aligned to the planning scene,

This is exactly the reason why the adapter exists.

> and the robot's reaction to that sort of mismatch should be to stop, not to attempt a potentially dangerous repair

I agree with the "potentially dangerous". The jiggle fraction is way to high by default, that's true. Feel free to look into it and reduce the default. Reducing the sampled jiggle to a maximum Cartesian displacement (of all links, not just the end effector) can also make sense.

Up to a temporary bug in the moveit_config template, time parameterization runs after this adapter, so it should not move faster than the rest of the trajectory.

The most prominent uses for this adapter is to get away from an uncertain table surface again after interacting with an object on top or to get out of contact with an object you do not accidentally want to pass through (i.e. allow collision checking).

rhaschke, 2020-08-20:
I agree that the jiggle fraction (which is in joint space) should be replaced by some absolute Cartesian error that is acceptable.
This error could be checked not only for the end effector but for all links of the robot.

simonschmeisser, 2020-09-07:
Anecdotally it sometimes also moves from inside a bin to the outside ... through the wall ... as there is no check for things not getting worse inbetween. I disabled it afterwards for our setups as that was with a grown up Kuka, not an iiwa.

When we ignore collisions like this, we check that the penetration depths does not exceed a certain limit on the way.

+1 for killing it and having people come up with better solutions (you want to have a cartesian movement in +z for that table example)

+0.5 for setting the factor WAY smaller or making it cartesian

rhaschke, 2024-02-02:
@Felix2M, I cannot reproduce your described behavior:
- I have the robot initially in collision with a scene object.
- I have set `jiggle_fraction` to zero, thus essentially disabling the `FixStartStateCollision` adapter.
- The adapter correctly reports: 
   `WARN 18:21:52 ros.moveit_ros_planning: Unable to find a valid state nearby the start state (using jiggle fraction of 0.000000 and 100 sampling attempts).`
- Planning correctly fails with:
  `WARN 18:21:52 ros.moveit_ros_planning_interface.move_group_interface: Fail: ABORTED: START_STATE_IN_COLLISION`

Felix2M, 2024-02-02:
@simonschmeisser, thank you for your suggestion. I will try that next week.

@rhaschke Yes, that is correct. I get this message too, but sometimes if the robot is stuck long enough this behavior occurs. Maybe it's because my robot is not stuck in a collision object or urdf link of the planning scene, but in quite small voxels of the octomap, which are artifacts of the zed 2i stereo camera.

rhaschke, 2024-02-02:
octomap obstactles should behave the same way as any other collision objects. Maybe your issue stems from the fact that the octomap (quickly) changes over time. Maybe we have race conditions?

Felix2M, 2024-02-02:
At the moment I am using a Yaskawa sia20f at 10% speed in a rather static environment and create the octomap with a view from the side to control the grasping process. So it's not a highly dynamic situation and as soon as the robot is stuck in some voxels it is completly static. I also don't know why these artifacts are not recognized by the self filtering as negligable. I increased the padding for the self filtering already that objects 20 cm away from the robot are not part of the octree anymore, but sometimes we get some voxels directly at the robotarm.

Is it possible that this problem occurs, because the padding and therefore the self filtering works different from the official documentation on our system?
The documention mentions: "padding_scale: Should always be >= 1.0". If we use a value >=1.0 the octomap does not show up. With a padding_scale of 0.001 to 0.02 it was working somehow. We previously thought that this was achieved by converting units.
For the DepthImageUpdater a padding_scale of >=1.0 is working as intended.

```
  - sensor_plugin: occupancy_map_monitor/PointCloudOctomapUpdater
    point_cloud_topic: /zed_multi/zed2i_staubli/zed_nodelet_staubli/point_cloud/pre_cloud_registered
    max_range: 5
    point_subsample: 1
    padding_offset: 0.2
    padding_scale: 0.005
    max_update_rate: 30
    filtered_cloud_topic: filtered_cloud
    ns: zed2i_staubli
```

As I am only working on that research project for two more weeks I will try to turn the FixStartStateCollision off to leave it in a secure condition. I will tell my colleagues to open a new issue in the future including a bag file. Thank you very much for your help!

rhaschke, 2024-02-02:
I cannot help with octomap issues unfortunately and refer to other colleagues, who hopefully chime in.
While you state that your setup is static, the depth camera dynamically yields new data all the time...