moveit/moveit #2281: boost serialization DSO missing

yonasteodros, 2020-08-27:
### Description

compiling from source giving an error '' libboost_serialization.so.1.65.1: error adding symbols: DSO missing from command line ''

### Your environment
* ROS Distro: [Melodic]
* OS Version: e.g. Ubuntu 18.04
* Source build
* origin/master branch

### Backtrace or Console output

`Errors     << moveit_planners_ompl:make /media/user/Data/ROS/ws_moveit/logs/moveit_planners_ompl/build.make.004.log
/usr/bin/ld: CMakeFiles/moveit_generate_state_database.dir/scripts/generate_state_database.cpp.o: undefined reference to symbol '_ZN5boost7archive6detail22archive_serializer_mapINS0_15binary_oarchiveEE6insertEPKNS1_16basic_serializerE'
//usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.65.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [/media/user/Data/ROS/ws_moveit/devel/.private/moveit_planners_ompl/lib/moveit_planners_ompl/generate_state_database] Error 1
make[1]: *** [ompl_interface/CMakeFiles/moveit_generate_state_database.dir/all] Error 2
make: *** [all] Error 2
cd /media/user/Data/ROS/ws_moveit/build/moveit_planners_ompl; catkin build --get-env moveit_planners_ompl | catkin env -si  /usr/bin/make --jobserver-fds=3,4 -j; cd -
`

v4hn, 2020-08-27:
Is that the same issue recently discussed in the moveit2 repository @henningkayser ?

henningkayser, 2020-09-15:
This error doesn't seem too familiar to me, but I wouldn't rule out that there are still some boost modules missing that could result in issues. However, `moveit_planners_ompl` didn't require additional modules.