moveit/moveit #2284: Add visual representation to CollisionObjects

felixvd, 2020-08-28:
Currently, `CollisionObject`s are purely collision shapes, which are usually primitive shapes for performance reasons. However, we don't want our scenes to look like Minecraft, so I want to add a visual representation to objects in the scene, so that we can display complex geometries like [this](https://user-images.githubusercontent.com/4535737/91532771-c37f3680-e949-11ea-8294-6c1ae6c48cc2.png) without calculating the collisions for every triangle. The robot and the environment objects already have this feature, but not the `CollisionObject`s that we add to the PlanningScene.

I would add the fields `visual_geometry_mesh` and `visual_geometry_pose` fields to the `CollisionObject` message, the `world` object and the `AttachedBody`. This is the simplest way and should be straight forward to implement. The only downside is that it might affect performance (visual mesh is kept in memory and copied with the `world`/`AttachedBody` objects) and that only one mesh can be used for visualization.

Regarding the restriction to a single mesh:  
I think this is fine. If we wanted to allow more visual geometries, this would be a much bigger job and break API, so I don't want to do it. It wouldn't be worth it. (outline: split [CollisionObject](http://docs.ros.org/melodic/api/moveit_msgs/html/msg/CollisionObject.html) message into visual and collision geometry ([like](https://github.com/ros-industrial-consortium/tesseract_ros/blob/master/tesseract_ros/tesseract_msgs/msg/VisualGeometry.msg) [this](https://github.com/ros-industrial-consortium/tesseract_ros/blob/master/tesseract_ros/tesseract_msgs/msg/CollisionGeometry.msg)), and in `world`/`AttachedBody`,  split `shapes_` and `shape_poses_` vectors into `visual` and `collision`).

Regarding performance:  
I *think* @rhaschke proposed (but I couldn't find where) to only pass in a mesh file location. That requires the mesh to be saved on disc and to be cached in `ros_visualization`. I would prefer not to implement it this way if it isn't necessary, since it might be premature optimization. Thoughts?

I looked at the visualization code, and using the `VisualVisible`/`CollisionVisible` flag to visualize the objects in `planning_scene_render.cpp` and `robot_state_visualization.cpp` won't be difficult. But I would like an opinion on the mesh vs mesh file consideration before I start. Thanks!

simonschmeisser, 2020-09-03:
Fully support that idea!

Considering the visualization, a filename/ resource url would be way faster as OGRE caches it. For collision meshes we currently reupload/reparse all on every geometry change in the planning scene (adding/removing/attaching/... any collision object). I have an open issue discussing this but cannot link it due to being on campground on mobile :)

A filename would also allow for fancy meshes (textures etc) in all formats supported by assimp.

I wouldn't be opposed to having both however for whatever dynamic mesh use-cases might arrise. (Could also be handled with immutable temp files however)

rhaschke, 2020-09-04:
I agree with @simonschmeisser, that a filename/resource url would be better.

v4hn, 2020-09-07:
> But I would like an opinion on the mesh vs mesh file consideration before I start.

I agree to filenames as well.
By core functionality nothing changes if the mesh files cannot be accessed for some reason, but adding the actual meshes in the message would create a lot of overhead. I just don't think that's worth it.

It should be possible to pass [`resource_retriever`](http://wiki.ros.org/resource_retriever) URLs though.

rhaschke, 2020-09-07:
> It should be possible to pass [`resource_retriever`](http://wiki.ros.org/resource_retriever) URLs though.

Sure!

felixvd, 2020-09-07:
Thanks for the input!

> Considering the visualization, a filename/ resource url would be way faster as OGRE caches it.

I assume the mesh is cached when we use [loadMeshFromResource](https://docs.ros.org/diamondback/api/rviz/html/namespacerviz.html#a7be90172209f6fc23df5de8447c7fa58), but that returns an `Ogre::MeshPtr`. The [renderShape](https://github.com/ros-planning/moveit/blob/master/moveit_ros/visualization/rviz_plugin_render_tools/src/render_shapes.cpp) function that's called for the [world objects](https://github.com/ros-planning/moveit/blob/master/moveit_ros/visualization/rviz_plugin_render_tools/src/planning_scene_render.cpp#L115) and [attached objects](https://github.com/ros-planning/moveit/blob/master/moveit_ros/visualization/rviz_plugin_render_tools/src/robot_state_visualization.cpp#L140) stores `rviz::Shape`s.

> For collision meshes we currently reupload/reparse all on every geometry change in the planning scene (adding/removing/attaching/... any collision object). I have an open issue discussing this but cannot link it due to being on campground on mobile :)

I guess this is what you meant and I feel like I remember you mentioning this more recently, but the only issue I could find about caching is https://github.com/ros-planning/moveit/issues/488 . Did you have anything in mind to improve this section of the code @simonschmeisser ? Is there another appropriate function to load the mesh? Reading it from disk at every loop seems nonsensical. I haven't looked at `rviz::Robot` or `rviz::Link` and I don't know where `render_shapes` is actually displayed, so this would save me some time.

rhaschke, 2020-09-07:
`rviz::Robot` handles both collision and visual geometries in the same fashion as mere Ogre `entities`:
https://github.com/ros-visualization/rviz/blob/2b31dc4be241e3ebd3a92c0f94d56fdb71391d51/src/rviz/robot/robot_link.cpp#L693-L694
https://github.com/ros-visualization/rviz/blob/2b31dc4be241e3ebd3a92c0f94d56fdb71391d51/src/rviz/robot/robot_link.cpp#L767-L768

`rviz::MeshShape` (used by MoveIt's `RenderShapes`) is designed to manually construct a mesh.
`RenderShapes` are not directly displayed by MoveIt's visualization. Rather, the corresponding `rviz::Shapes` (and their Ogre `entities`) are created once via [`renderShape()`](https://github.com/ros-planning/moveit/blob/master/moveit_ros/visualization/rviz_plugin_render_tools/src/render_shapes.cpp) and stored in the Ogre scene. Hope that helps.

simonschmeisser, 2020-09-07:
Sorry for not spending the time to look that up ... #2079 is the one about the mesh msgs of CollisionObjects being translated into Ogre data structures too often in PlanningSceneDisplay

felixvd, 2020-09-08:
> `rviz::Robot` handles both collision and visual geometries in the same fashion as mere Ogre `entities`: [...]

Thanks. 

When I [made a prototype last week](https://github.com/felixvd/moveit/commit/4fa15ed980ec1d20304104ad385ab21bb7bf2c36), I noticed that the objects are drawn without considering the Visual/Collision switch, so I added the logic to [planning_scene_render](https://github.com/felixvd/moveit/commit/4fa15ed980ec1d20304104ad385ab21bb7bf2c36#diff-8f1dc572b985dca832c536bd7ae3ff7cR132) and [robot_state_visualization](https://github.com/felixvd/moveit/commit/4fa15ed980ec1d20304104ad385ab21bb7bf2c36#diff-4fda7e54aaa6ae32df4b027d40473526R142), but it seems redundant. Is there a more sensible way, e.g. to inherit from `rviz::RobotLink` for the Visual/Collision distinction? Or should we just store the geometry in planning_scene_render and robot_state_visualization?

> `rviz::MeshShape` (used by MoveIt's `RenderShapes`) is designed to manually construct a mesh.
> `RenderShapes` are not directly displayed by MoveIt's visualization. Rather, the corresponding [...]

Those meshes are created from scratch every time, right? Is this where you would have implemented Variant B you described in #2079 @simonschmeisser ?

felixvd, 2021-03-10:
I made progress during this WMD: 

[image]

But it looks like I might have to deal with some more Rviz/Ogre details, and maybe split up the `Ogre::Scene` in the `planning_scene_display`.

The split between robot and scene geometry is somewhat tiring. I kind of wish we had a simpler planning scene.

simonschmeisser, 2021-03-10:
Nice, what branch would that be in? Or would it make sense to open a WIP PR? I might be able to help a bit with the ogre/visualization part