moveit/moveit #2285: Should we use M_PI or boost::math::constants::pi<double>() ?

JeroenDM, 2020-08-30:
They are used both, somethings in the same source file. For example `revolute_joint_model.cpp` uses boost pi on [this line](https://github.com/ros-planning/moveit/blob/7ad2bc7b86dad08061d98668ba34feba54bb05cc/moveit_core/robot_model/src/revolute_joint_model.cpp#L155) and just a couple of lines further on, `M_PI` is used.

A quick search on Github gives 15 occurrences of `M_PI` and 13 occurrences of `boost::math::constants::pi<double>()`. So cleaning it up would be easy.

Personally I feel that `M_PI` is much more readable. But there could be other issues I'm not aware of. If we have to use `boost::math::constants::pi<double>()`, I would still propose to make it into a const to make the code easier to reason about.

I'm sorry if I missed any earlier discussion on this, I could not find one.

rhaschke, 2020-08-30:
I guess this is a matter of taste and depends on the contributor. I would prefer `M_PI` for readability as well.

gavanderhoorn, 2020-08-30:
From a C++ perspective, using an untyped macro defined in a C header isn't necessarily the best idea.

If readability is important, aliasing the namespace could be an option (`namespace bmc = boost::math::constants`) or even creating a utility header with often used constants (`static constexpr pi = bmc::pi<double>()`) (pseudo-code, haven't checked).

JeroenDM, 2020-08-31:
@gavanderhoorn I don't know how I missed it, but the [MoveIt Style Guidelines](https://moveit.ros.org/documentation/contributing/code/) already agree with your point:
> Avoid C-style functions such as FLT_EPSILON - instead use std::numeric_limits<double>::epsilon()

So indeed we could go for `static constexpr double PI{ boost::math::constants::pi<double>() };` inside the namespace of the file where it is used. Then we can avoid code such as:

```C++
    {
      if (diff > 0.0)
        diff = 2.0 * boost::math::constants::pi<double>() - diff;
      else
        diff = -2.0 * boost::math::constants::pi<double>() - diff;
      state[0] = from[0] - diff * t;
      // input states are within bounds, so the following check is sufficient
      if (state[0] > boost::math::constants::pi<double>())
        state[0] -= 2.0 * boost::math::constants::pi<double>();
      else if (state[0] < -boost::math::constants::pi<double>())
        state[0] += 2.0 * boost::math::constants::pi<double>();
    }
```

I know it's only I minor nit, but it really stood out when reading through the code....

gavanderhoorn, 2020-08-31:
Not saying we should use it, but [levmin/P0631](https://github.com/levmin/P0631) provides a reference implementation of this proposal: [Math Constants](http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2019/p0631r8.pdf).

It's likely future C++ standards will get something *like this*.

JeroenDM, 2020-08-31:
> And C++20 has the numbers header.

Ah, nice! I certainly will be using that one in the future :)