moveit/moveit #2286: Robot not displayed in setup assistant Melodic

ammaar8, 2020-08-31:
### Description
Hello,
I tried to use moveit setup assistant with open manipulator for learning but after loading it didn't show up in the display so I tried it with the franka_ros description and have the same issue. I saw similar issue for ros-indgo and did try exporting ` export $LC_NUMERIC` but it didn't change anything.

### Your environment
* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.04
* Source or Binary build? Binary
* If binary, which release version? 1.0.6-Alpha

### Steps to reproduce
I cloned the frank_ros description while following the setup-assistant tutorial on https://ros-planning.github.io/ and loaded the panda_arm_hand.urdf.xacro file. The setup happens and all the collisions are shown in the second tab but it is not displayed in the visual panel.

### Backtrace or Console output
[backtrace](https://gist.github.com/ammaar8/9dba184c60ad51b88170366acd546307)

felixvd, 2020-09-01:
What command are you running exactly? Does the display for `roslaunch panda_moveit_config demo.launch` work?

ammaar8, 2020-09-01:
I installed the franka_ros package using `sudo apt-get install ros-melodic-franka-description` 
then ran `roslaunch moveit_setup_assistant setup_assistant.launch`.
After selecting the .urdf.xacro I get this - 
[image]

Yes  `roslaunch panda_moveit_config demo.launch` works fine

rhaschke, 2020-09-01:
I can confirm this bug. rviz's RobotModel correctly displays the robot. Thus, seems to be related to MSA. I will look into this asap.

rhaschke, 2020-09-01:
The problem seems to be fixed in MoveIt's master branch.

rhaschke, 2020-09-01:
The culprit was backport #2245, which partially included #1625, but not (yet) its fixup #1738. Should be fixed in melodic-devel now.

closed 2020-09-01 by rhaschke

brandonlawrence1996, 2020-10-01:
@rhaschke i am still facing the issue where the robot model doesn't load in the setup assistant. May I know what is the fix to this problem?

rhaschke, 2020-10-01:
As we didn't yet re-released in Melodic, you need to build melodic-devel branch from source to get this fix.

GeorgeBulimar, 2021-09-27:
Hi,

@rhaschke I'm facing the same issue. What can I do to fix it? :)

rhaschke, 2021-09-28:
You need to build melodic-devel branch from source. Follow the instructions [here](https://moveit.ros.org/install/source).

rhaschke, 2021-09-28:
@GeorgeBulimar, I noticed that this issue is more than a year old. Since then, of course we had new releases of MoveIt into Melodic and Noetic. For me, the robot model is correctly displayed in MSA.
If you still have issues, please file a new issue report also specifying your MoveIt version and environment.