moveit/moveit #2289: Rviz FPS drops to zero with controller plugged in and with "allow external comm." enabled.

RealWilliamWells, 2020-09-03:
### Description

Rviz FPS drops to zero with controller plugged in and with "allow external comm." enabled, when using joystick controls. The FPS returns to normal upon unplugging the controller or disabling "allow external comm.". This same issue occurs with the UR3 Arm from https://github.com/PUT-UGV-Team/ur3_erc_docker.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Binary obtained with `apt-get install`
* Package: ros-melodic-moveit
* Version: 1.0.6-1bionic.20200822.021945

### Steps to reproduce
1. Install MoveIt on ROS Melodic on Ubuntu 18.04.
2. Create catkin workspace and run `git clone https://github.com/ros-planning/panda_moveit_config.git` to clone Panda Arm.
3. Build everything in catkin workspace with `catkin_make`. Source "devel/setup.bash". Setup Panda arm for operation in step one on the Panda Arm tutorial: https://rosplanning.github.io/moveit_tutorials/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html.
4. Plug in controller run `roslaunch panda_moveit_config joystick_control.launch` and enable "allow external comm."
5. Notice how the FPS in Rviz drops to very low single digits or zero.

### Expected behaviour
FPS should remain stable and allow the controlling of the arm.

### Actual behaviour
FPS drops to single digits or zero, making the arm nearly uncontrollable.

### Backtrace or Console output
You can start planning now!

**part of demo.launch output**
```
[ERROR] [1599091377.251573909]: PluginlibFactory: The plugin for class 'rviz_visual_tools/KeyTool' failed to load.  Error: According to the loaded plugin descriptions the class rviz_visual_tools/KeyTool with base class type rviz::Tool does not exist. Declared types are  rviz/FocusCamera rviz/Interact rviz/Measure rviz/MoveCamera rviz/PublishPoint rviz/Select rviz/SetGoal rviz/SetInitialPose rviz_plugin_tutorials/PlantFlag
[rospack] Error: no package given
[librospack]: error while executing command
[rospack] Error: no package given
[librospack]: error while executing command
[ERROR] [1599091377.269937288]: PluginlibFactory: The plugin for class 'rviz_visual_tools/RvizVisualToolsGui' failed to load.  Error: According to the loaded plugin descriptions the class rviz_visual_tools/RvizVisualToolsGui with base class type rviz::Panel does not exist. Declared types are  rviz_plugin_tutorials/Teleop
```

**part of joystick_control.launch output**
```
process[joy-1]: started with pid [25677]
process[moveit_joy-2]: started with pid [25678]
[ WARN] [1599091619.137128709]: Couldn't set gain on joystick force feedback: Bad file descriptor
[ INFO] [1599091619.837621421]: Loading robot model 'panda'...
panda_arm_hand ['/rviz/moveit/move_marker/goal_panda_link8']
panda_arm ['/rviz/moveit/move_marker/goal_panda_link8']
```

MartensCedric, 2020-09-03:
I have the same issue with the UR3 robotic arm so it doesn't seem specific to the Panda arm

gavanderhoorn, 2020-09-03:
@RealWilliamWells: just an observation: you appear to be using `ur_modern_driver` in your `Dockerfile`. Are you aware of the fact that it is completely deprecated and you should use [Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) instead?

Unless you have a CB2 controller, that is.

RealWilliamWells, 2020-09-03:
> @RealWilliamWells / @MartensCedric: just an observation: you appear to be using `ur_modern_driver` in your `Dockerfile`. Are you aware of the fact that it is completely deprecated and you should use [Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) instead?
> 
> Unless you have a CB2 controller, that is.

Oh I wasn't aware actually thanks! I just tested with `Universal_Robots_ROS_Driver` instead, and it produced the same results.

gavanderhoorn, 2020-09-03:
> Oh I wasn't aware actually thanks

[The notice](https://github.com/ros-industrial/ur_modern_driver#deprecation-notice) is rather hard to miss:

> ## Deprecation notice
> 
> note: the `master` branch contains a deprecated version of this driver, kept for archival purposes only.
> 
> `ur_modern_driver` has been deprecated completely in favour of [ur_robot_driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver).
> 
> Users with CB3 and e-Series controllers should use `ur_robot_driver`.
> 
> Refer to the [announcement on ROS Discourse](https://discourse.ros.org/t/deprecation-of-ur-modern-driver/11629) for more information.

Besides that the package manifest also marks it as `deprecated`, so you should see warnings when building your workspace.

> I just tested with `Universal_Robots_ROS_Driver` instead, and it produced the same results.

I did not expect anything to change.

simonschmeisser, 2020-09-07:
Do you have `rviz_visual_tools` installed? That should solve the first error

For the second, something seems to be wrong with your joystick?

RealWilliamWells, 2020-09-09:
Hey thanks for the reply! I was able to solve the first error by installing `rviz_visual_tools`, but I wasn't able to solve the second one even when using a different controller. The FPS issue with Rviz still persists even after solving the first error.

rhaschke, 2020-09-09:
I cannot reproduce the second issue, because I don't have access to a joystick... Without a joystick attached, there is not drop in performance when _allowing external comm_.

MartensCedric, 2020-09-10:
> I cannot reproduce the second issue, because I don't have access to a joystick... Without a joystick attached, there is not drop in performance when _allowing external comm_.

Yes this is the same behaviour. But the moment a joystick is plugged the FPS goes to 0-1 FPS, but once unplugged it goes back to 30+. Really weird

felixvd, 2020-10-15:
If you could try `rostopic echo`ing [these topics](https://github.com/ros-planning/moveit/blob/master/moveit_ros/visualization/motion_planning_rviz_plugin/src/motion_planning_frame.cpp#L267) and the topics under `/rviz/moveit/move_marker/` when you plug in your joystick, that would help debug this issue.

Sunshengjin, 2020-11-26:
Hi, I have same problem ..WHen i running xxx_moveit_config/launch/joystick_control.launch , rviz FPS drop to 0.. 
Is anyone have any solutions?

gy2256, 2021-02-22:
I have the same issue. The rviz is so laggy and the entire things become not usable. I thought it was taking too long to run the IK solver but apparently that was not the issue.

gy2256, 2021-02-27:
I work around the problem by directly controlling the manipulator in gazebo environment. It works ok.

Mad0Scientisto, 2021-03-15:
I also had the same problem: I tried the official MoveIt tutorial, in particular the Joystick tutorial (https://ros-planning.github.io/moveit_tutorials/doc/joystick_control_teleoperation/joystick_control_teleoperation_tutorial.html) with noetic ROS on a native Ubuntu machine: after connecting the joystick and enabling "External comm." the fps dropped to 0. After some testing, I tried removing lines 849 and 850 of the file `moveit/moveit_ros/visualization/src/moveit_ros_visualization/moveitjoy_module.py`

`if self.counter% 10:`
`             self.update_start_state_pub.publish(Empty())`

Now the joystick seems to work at full fps. However, I have no idea about the usefulness of that publisher.