moveit/moveit #2290: How to add multiply collision objects in the scene(worlld) ?

astronaut71, 2020-09-03:
Hi

I would like to add multiply collision objects in Moveit Rviz. I detected the objects (chairs) with darknet_ros and able to get the bounding box of the objects (chairs). When trying to add them as collision objects in Moveit it only able to add one, but I have detected two chairs and want to add two collision objects in the scene. In Rviz Im able to add and visualize only one, but in the terminal output there are two.

My environment

    ROS Distro: Kinetic
    OS Version: e.g. Ubuntu 16.04

Here the Code with the Moveit interface

```
using namespace sensor_msgs;
using namespace std;

void chatterCallback(const darknet_ros_3d_msgs::BoundingBoxes3d::ConstPtr& msg)
{ 
	       cout<<"Entering call back function" <<endl;
           int person_count=0;
           int inc = 1;
           while (msg->bounding_boxes[0].Class == "chair")
						{
                         
    			          	for(int count=0; count <= chair_count; count++)
								{
								

								const std::string PLANNING_GROUP = "crane_control";
								moveit::planning_interface::MoveGroupInterface move_group(PLANNING_GROUP);
								moveit::planning_interface::PlanningSceneInterface planning_scene_interface;

								string str= to_string(count);
								moveit_msgs::CollisionObject collision_object;
								collision_object.header.frame_id = "world";
								collision_object.id = "BOX_chair__" + str;
								cout<<"Chair:" << collision_object.id <<endl;
	
								shape_msgs::SolidPrimitive primitive;
								primitive.type = primitive.BOX;
								primitive.dimensions.resize(3);

								std::vector<float> array_xdim(inc);
								std::vector<float> array_ydim(inc);
								std::vector<float> array_zdim(inc);
					
								array_xdim[inc] = msg->bounding_boxes[0].xmax - msg->bounding_boxes[0].xmin;
								array_ydim[inc] = msg->bounding_boxes[0].ymax - msg->bounding_boxes[0].ymin;
								array_zdim[inc] = msg->bounding_boxes[0].zmax - msg->bounding_boxes[0].zmin;
								cout<<"dimx:" <<  array_xdim[inc] <<endl;
								cout<<"dimy:" <<  array_ydim[inc] <<endl;
								cout<<"dimz:" <<  array_zdim[inc] <<endl;

								primitive.dimensions[0] = array_xdim[inc];
								primitive.dimensions[1] = array_ydim[inc];
								primitive.dimensions[2] = array_zdim[inc];   
								geometry_msgs::Pose box_pose;

								std::vector<float> array_xpos(inc);
								std::vector<float> array_ypos(inc);
								std::vector<float> array_zpos(inc);
	
								array_xpos[inc] = (msg->bounding_boxes[0].xmax + msg->bounding_boxes[0].xmin)/2;
								array_ypos[inc] = (msg->bounding_boxes[0].ymax + msg->bounding_boxes[0].ymin)/2;
								array_zpos[inc] = (msg->bounding_boxes[0].zmax + msg->bounding_boxes[0].zmin)/2;

								box_pose.position.x = array_xpos[inc];
								box_pose.position.y = array_ypos[inc];
								box_pose.position.z = array_zpos[inc];
								box_pose.orientation.w = 1.0;	
	
								cout<<"position_x:" <<  box_pose.position.x <<endl;
								cout<<"position_y:" <<  box_pose.position.y <<endl;
								cout<<"position_z:" <<  box_pose.position.z <<endl;

								collision_object.primitives.push_back(primitive);
								collision_object.primitive_poses.push_back(box_pose);
								collision_object.operation = collision_object.ADD;
								std::vector<moveit_msgs::CollisionObject> collision_vector;
								collision_vector.push_back(collision_object);
								planning_scene_interface.applyCollisionObjects(collision_vector);
								cout<<"Chair count" << chair_count <<endl;
								}
   	 				chair_count++;
                    inc++;
  					}
              cout<<"Exiting call back function" <<endl;
}

int main(int argc,  char** argv)
{
  ros::init(argc, argv, "cpp_subscriber");
  ros::NodeHandle n;
  ros::Subscriber sub = n.subscribe("/darknet_ros_3d/bounding_boxes", 200, chatterCallback);
  ros::spin();	
  return 0;

}
```

Here the Rviz image
[image]

And here the terminal output
[image]

Any help?

rhaschke, 2020-09-03:
You processing code is broken: You are using 3 counters: `inc`, `count`, and `chair_count` in a way that is not clear at all.
But, most importantly, you only access the first bounding box: `msg->bounding_boxes[0]`. Thus, all collision objects are probably created in the same position. You can easily check the collision objects on the tab `Scene Objects` of the `Motion Planning` plugin.

closed 2020-09-03 by rhaschke

gavanderhoorn, 2020-09-03:
Also on ROS Answers: [Not able to add multiply collision objects in Moveit Rviz](https://answers.ros.org/question/360731).

astronaut71, 2020-09-03:
ok yes it should be while (msg->bounding_boxes[inc].Class == "chair")   and msg->bounding_boxes[inc] But any other help? You can easily check the collision objects on the tab Scene Objects of the Motion Planning plugin. What do you mean?

astronaut71, 2020-09-03:
> You processing code is broken: You are using 3 counters: `inc`, `count`, and `chair_count` in a way that is not clear at all.
> But, most importantly, you only access the first bounding box: `msg->bounding_boxes[0]`. Thus, all collision objects are probably created in the same position. You can easily check the collision objects on the tab `Scene Objects` of the `Motion Planning` plugin. But any help ?

felixvd, 2020-09-03:
This issue board is for bugs in and development of MoveIt. For help on how to use MoveIt and ROS, answers.ros.org is the appropriate place. 

> You can easily check the collision objects on the tab Scene Objects of the Motion Planning plugin. What do you mean?

Unfold the Motion Planning Plugin in the Displays panel on the top left in your Rviz screenshots, and you will find the collision objects listed in there.

astronaut71, 2020-09-03:
I dont have the Scene Objects tab  of the Motion Planning plugin. I have Scene Robot and Scene Geometry tabs and others but not Scene Objects

felixvd, 2020-09-03:
I should sleep more. The Scene Objects tab of the plugin is not in the displays but here:

[image]