moveit/moveit #2292: MSA: cancel components dialog for new planning group

v4hn, 2020-09-03:
In the "Planning Groups" tab of the MoveIt Setup Assistant, users can add new planning groups to their configuration:

[image]

As the last step of this process, the user should click any of the four buttons on the bottom to assign an actual structure to the robot.
(Someone should really add the "Add Kin. Chain" button to "Recommended" as well. It's the most generic/user-friendly way to describe simple arms..)

Clicking on any of the buttons opens a new form to assign the structures with a "Save" and "Cancel" button on the bottom:

[image]

Pressing the "Cancel" button should revert the dialog to the previous dialog, but it actually cancels the whole process to add a new group, throwing away the information entered already.

beingkartik, 2020-09-09:
Hey, I would like to take this up. However, this will be my first time working on Moveit issues and need some help getting started. I guess we need to make the changes [here](https://github.com/ros-planning/moveit/tree/4611219b472719d8163b856c8fbc6f9eed09b2a4/moveit_setup_assistant/src). But I can't find this in the moveit installation in my system. Do I need to build from source to be able to do that? I currently have it installed using sudo apt-get install ros-kinetic-moveit.

rhaschke, 2020-09-09:
Yes, you need to build from source. But you can limit your build to moveit_setup_assistant only, e.g. using catkin tools like this:
`catkin build --no-deps moveit_setup_assistant`

It would be great if you could provide a patch against the Melodic branch though.

PR #2350 (2020-10-07): MSA: Fix group editing
Fix #2292. Note, that this builds on top of #2349!