moveit/moveit #2296: White paper/references with pseudo code for the iterative spline parametrization

karshtharyani, 2020-09-04:
I have been trying to understand some of the drawbacks of the parameterization strategy here. Is there a white paper or reference with some pseudo-code explaining what is happening here?

At a high-level I understand stretching of the time intervals to attain max_velocity and max_acceleration and fitting a smooth curve, but don't quite grasp the implementation details here(especially the for `atfactor` updates).

Also, I think these two links somewhere in the source code will be helpful for someone stale on cubic-splines and tridiagonal substitution. 

[Tridiagonal Matrix Algorithm](https://en.wikipedia.org/wiki/Tridiagonal_matrix_algorithm)
[Clamped Cubic Splines](http://cis.poly.edu/~mleung/CS3734/s03/ch07/cubicSpline.pdf)

```
while (1)
  {
    int loop = 0;

    // Calculate the interval stretches due to acceleration
    std::vector<double> time_factor(num_points - 1, 1.00);
    for (unsigned j = 0; j < num_joints; j++)
    {
      // Move points to satisfy initial/final acceleration
      if (add_points_)
      {
        adjust_two_positions(num_points, &time_diff[0], &t2[j].positions_[0], &t2[j].velocities_[0],
                             &t2[j].accelerations_[0], t2[j].initial_acceleration_, t2[j].final_acceleration_);
      }

      fit_cubic_spline(num_points, &time_diff[0], &t2[j].positions_[0], &t2[j].velocities_[0], &t2[j].accelerations_[0]);
      for (unsigned i = 0; i < num_points; i++)
      {
        const double acc = t2[j].accelerations_[i];
        double atfactor = 1.0;
        if (acc > t2[j].max_acceleration_)
          atfactor = sqrt(acc / t2[j].max_acceleration_);
        if (acc < t2[j].min_acceleration_)
          atfactor = sqrt(acc / t2[j].min_acceleration_);
        if (atfactor > 1.01)  // within 1%
          loop = 1;
        atfactor = (atfactor - 1.0) / 16.0 + 1.0;  // 1/16th
        if (i > 0)
          time_factor[i - 1] = std::max(time_factor[i - 1], atfactor);
        if (i < num_points - 1)
          time_factor[i] = std::max(time_factor[i], atfactor);
      }
    }
```

gavanderhoorn, 2020-09-04:
Afaik there is no resource such as you ask for.

This was written a very long time ago by @kennethanderson (as is the spline-based version), but other than the comments in the files I don't believe there is any additional documentation.

felixvd, 2020-09-04:
Personally I'm in favor of more in-line documentation at this level of detail (although "Use tridiagonal matrix algorithm" is surely more sensible than a wikipedia link), so I'd welcome a PR. If you make a good summary for yourself that you would like to share, it can live in the [developer concepts](https://moveit.ros.org/documentation/concepts/developer_concepts/) as well.