moveit/moveit #2297: Moving code that depends on QT out of the MoveIt repo

tylerjw, 2020-09-04:
As much of our compile time is as a result of qt elements it would be nice to move those into a separate repo.  The advantages I see in doing this are:

* Less untested code in the moveit repo (we don't currently test any gui elements)
* People who use moveit without qt won't have to install that dependency
* Speed up CI

Disadvantages:

* It would break existing setups that depend on master (would need to give users warning)
* It would be a large project (hours spent) that wouldn't increase the capabilities of moveit
* PRs that change QT code and/or the interface to that code will be made more difficult

I'm posting this to see what other's thoughts are on this.

_Tangent_: If we move in the direction move repos we should consider work on CI infrastructure for testing downstream repos we control in some way to catch when a change would break something downstream.  I'm not proposing breaking up all of moveit, just moving the untested qt code out, so downstream testing doesn't help when what is downstream doesn't have tests.

mikeferguson, 2020-09-04:
A possible alternative would be a script that places appropriate CATKIN_IGNORES for a "headless" build without QT - they are already separate ROS packages, right? Or has QT leaked into things like moveit_core over time?

> _Tangent_: If we move in the direction move repos we should consider work on CI infrastructure for testing downstream repos we control in some way to catch when a change would break something downstream

I think the MoveIt CI is based on Industrial CI, right? Industrial CI already supports testing downstream repos, FYI.

Final thought: perhaps do this in MoveIt2 and then decide whether to also do it in ROS1?

tylerjw, 2020-09-04:
> I think the MoveIt CI is based on Industrial CI, right? Industrial CI already supports testing downstream repos, FYI.

They seem to be loosely based on each other a long time ago.  The code has diverged quite a bit and we are not using industrial_ci here.  That would be one feature we would gain by moving to industrial_ci.

> Final thought: perhaps do this in MoveIt2 and then decide whether to also do it in ROS1?

Good point.  That is probably the best approach to any large restructuring efforts anyway.

gavanderhoorn, 2020-09-04:
@mikeferguson wrote:
> A possible alternative would be a script that places appropriate CATKIN_IGNORES for a "headless" build without QT

Or explicitly list the packages which should / shouldn't be built?

Both Catkin and Colcon support that.

Wouldn't require splitting up nor fiddling with `CATKIN_IGNORE` files.

rhaschke, 2020-09-05:
This approach counteracts previous approaches to merge closely related packages into a single repo.
Using ccache, building time is not the major issue for Travis, but testing is!
What about a mechanism to limit tests to packages that have seen changes in a PR?

tylerjw, 2020-09-05:
In this build: https://travis-ci.com/github/ros-planning/moveit/jobs/381214917

Even with ccache the build time is 220s and the test time is 268s.  I think there are probably things we should do to decrease test time (like fixing those tests I wrote that use normal execution to use the jump to the last position execution) but disabling them will make it take even longer to find the bugs like the one I'm trying to figure out that is causing that job to fail.

tylerjw, 2020-09-05:
This was just an idea for how we could reduce the CI time (it does have real drawbacks and probably isn't worth it).

closed 2020-09-05 by tylerjw