moveit/moveit #2298: Tests failing for moveit_core on Noetic in Travis

tylerjw, 2020-09-04:
### Description

Here is the travis run: https://travis-ci.com/github/ros-planning/moveit/jobs/381214917

I'm going to work on reproducing this locally and try doing a git bisect of recent merges on dependencies to see if I can find the source.  Nothing was merged into master here since this started failing.  Any help would be appreciated.

tylerjw, 2020-09-04:
```<failure message="unknown file&#x0A;C++ exception with description &quot;Unable to update the SRDF Model&quot; thrown in the test body." type=""><![CDATA[unknown file

C++ exception with description "Unable to update the SRDF Model" thrown in the test body.]]></failure>```

tylerjw, 2020-09-04:
```
-- run_tests.py: execute commands
  /home/tyler/workspace/ws_moveit_noetic/devel/.private/moveit_core/lib/moveit_core/test_aabb --gtest_output=xml:/home/tyler/workspace/ws_moveit_noetic/build/moveit_core/test_results/moveit_core/gtest-test_aabb.xml
[==========] Running 3 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 3 tests from TestAABB
[ RUN      ] TestAABB.TestPR2
ros.moveit_core.robot_model: Loading robot model 'pr2'...
ros.moveit_core.robot_model.empty_collision_geometry: Link sensor_mount_link has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
ros.moveit_core.robot_model.empty_collision_geometry: Link double_stereo_link has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
ros.moveit_core.robot_model: Group state 'tuck_arms' doesn't specify all group joints in group 'arms'. l_shoulder_lift_joint, l_upper_arm_roll_joint, l_elbow_flex_joint, l_forearm_roll_joint, l_wrist_flex_joint, l_wrist_roll_joint, r_shoulder_pan_joint, r_shoulder_lift_joint, r_upper_arm_roll_joint, r_elbow_flex_joint, r_forearm_roll_joint, r_wrist_flex_joint, r_wrist_roll_joint are missing.
[       OK ] TestAABB.TestPR2 (63 ms)
[ RUN      ] TestAABB.TestSimple
Error:   Could not find the 'robot' element in the xml file
         at line 638 in /tmp/binarydeb/ros-noetic-srdfdom-0.6.0/src/model.cpp
unknown file: Failure
C++ exception with description "Unable to update the SRDF Model" thrown in the test body.
[  FAILED  ] TestAABB.TestSimple (0 ms)
[ RUN      ] TestAABB.TestComplex
Error:   Could not find the 'robot' element in the xml file
         at line 638 in /tmp/binarydeb/ros-noetic-srdfdom-0.6.0/src/model.cpp
unknown file: Failure
C++ exception with description "Unable to update the SRDF Model" thrown in the test body.
[  FAILED  ] TestAABB.TestComplex (0 ms)
[----------] 3 tests from TestAABB (63 ms total)

[----------] Global test environment tear-down
[==========] 3 tests from 1 test suite ran. (63 ms total)
[  PASSED  ] 1 test.
[  FAILED  ] 2 tests, listed below:
[  FAILED  ] TestAABB.TestSimple
[  FAILED  ] TestAABB.TestComplex
```

tylerjw, 2020-09-04:
```
#0  0x0000000004ae0762 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00000000050a36aa in ?? () from /opt/ros/noetic/lib/libsrdfdom.so.0.6.0
#2  0x0000000004947cbf in moveit::core::RobotModelBuilder::build (this=<optimized out>)
    at /usr/include/c++/9/bits/shared_ptr_base.h:1020
#3  0x0000000000115ece in TestAABB_TestSimple_Test::TestBody (this=<optimized out>)
    at /home/tyler/workspace/ws_moveit_noetic/src/moveit/moveit_core/robot_state/test/test_aabb.cpp:291
#4  0x000000000489e8f1 in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void> (
    location=0x48a0b9a "the test body", method=<optimized out>, object=0xa0e22d0)
    at /usr/src/googletest/googletest/src/gtest.cc:2414
#5  testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void> (object=object@entry=0xa0e22d0, 
    method=<optimized out>, location=location@entry=0x48a0b9a "the test body")
    at /usr/src/googletest/googletest/src/gtest.cc:2469
#6  0x0000000004891c36 in testing::Test::Run (this=0xa0e22d0) at /usr/src/googletest/googletest/src/gtest.cc:2508
#7  testing::Test::Run (this=0xa0e22d0) at /usr/src/googletest/googletest/src/gtest.cc:2498
#8  0x0000000004891d95 in testing::TestInfo::Run (this=0x9dae2f0) at /usr/src/googletest/googletest/src/gtest.cc:2684
#9  testing::TestInfo::Run (this=0x9dae2f0) at /usr/src/googletest/googletest/src/gtest.cc:2657
#10 0x0000000004891e7d in testing::TestSuite::Run (this=0x9dade30) at /usr/src/googletest/googletest/src/gtest.cc:2816
#11 testing::TestSuite::Run (this=0x9dade30) at /usr/src/googletest/googletest/src/gtest.cc:2795
#12 0x000000000489239c in testing::internal::UnitTestImpl::RunAllTests (this=0x9dad830)
    at /usr/include/c++/9/bits/stl_vector.h:1040
#13 0x000000000489ee61 in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (location=0x48a1fb0 "auxiliary test code (environments or event listeners)", method=<optimized out>, 
    object=0x9dad830) at /usr/src/googletest/googletest/src/gtest.cc:2414
```

tylerjw, 2020-09-04:
My suspicion is that this is caused by the robot model builder's call into libsrdfdom and that somehow we are now hitting this line: https://github.com/dlutguobin/RC/blob/9bdc2c08e9b29e514ff2858726b1ea5f44fb6ffe/srdfdom/src/srdf_writer.cpp#L111

I'm going to build with debug symbols to see if I can figure it out.  Also, did something change in the srdfdom project from melodic to noetic that might be causing this?

tylerjw, 2020-09-04:
I checked out srdfdom and the problem went away on the melodic-devel branch.  That means it probably has something todo with the tinyxml2 library port.

tylerjw, 2020-09-05:
Using gdb I've found that this block of code:

https://github.com/ros-planning/srdfdom/blob/6a9c757f0e8b5268c0887ecc396b140de7add775/src/model.cpp#L627-L634

Is not doing what it used to.  With the old library `FindFirstChildElement("robot")` returns a non-zero result and in this case it returns 0.  Here is some gdb output from breaking inside the if statement (on the debug print):

```c++
(gdb) print xml
$9 = (tinyxml2::XMLDocument *) 0x1ffeffe230
(gdb) print robot_xml
$10 = (tinyxml2::XMLElement *) 0x0
(gdb) print xml->FirstChildElement("robot")
Couldn't find method tinyxml2::XMLDocument::FirstChildElement
```

_I am frustrated by the fact that I can't seem to get any symbol data or info on objects inside the tinyxml library (I am building in debug mode)._

tylerjw, 2020-09-05:
@rhaschke do you have any thoughts or advice on this?  I'm frustrated by gdb's lack of symbols for the tinyxml library but I think I figured out the exact place where the two seem to differ.  I'm not sure how to ask why they differ at this point.  Not being able to look inside the tiny xml object is making this difficult.  I've gone around looking at the variables I can see in the model.cpp and for example the xmlstring in both cases is the same (I posted it above) which means that the same data is being passed in to tinyxml in the method "Parse" on line 664 in the old one and 670 in the new one in `model.cpp`.  Both of those pass the next check (they don't error in Parse) and they return the initXml call where we find this line.

rhaschke, 2020-09-06:
Thanks for your detailed analysis. I will have a look into this right now.

rhaschke, 2020-09-06:
The culprit was the SRDFWriter, which failed to populate the XML document with the root <robot> tag.
I fixed this issue locally. I am adding a unittest to srdfdom now and will re-relase subsequently.

rhaschke, 2020-09-06:
srdfdom release request opened: https://github.com/ros/rosdistro/pull/26470

closed 2020-09-06 by rhaschke

tylerjw, 2020-09-06:
Sweet, thank you for fixing this.  I didn't really know what I was looking for (other than it not working the way it used to).

VictoriaBentell, 2020-09-08:
@rhaschke If it's helpful at all, I'm encountering a similar issue using moveit_setup_assistant. I'm using the [UR5 urdf file](https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/ur_description/urdf/ur5_robot.urdf.xacro) to create a new configuration package, but just after adding a planning group (tested with Add Joints, Add Links, & Add Kin. Chain) and hitting Save, the window crashes and I get `[ERROR] [1599544434.976813810]: Could not find the 'robot' element in the xml file
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to update the SRDF Model`
The same problem occurs with my own custom robot as well, but I doubt that posting the code for that will provide much insight.
(I used the noetic_devel branch, installed from source a couple hours ago.)

rhaschke, 2020-09-08:
@VictoriaBentell, please try to update to the latest release of srdfdom (which is only available via [ros-testing](http://wiki.ros.org/TestingRepository) right now, because it's not yet synced.). Alternatively, you can build this package from source.

VictoriaBentell, 2020-09-08:
Fixed after reinstalling MoveIt! from ros-testing.

peci1, 2021-03-30:
@tylerjw I was also debugging a tinyxml2 -related issue today and I was also sad about not having the debug symbols, even though I installed package `libtinyxml2.6.2v5-dbg`. Surprise, these are debugging symbols for tinyxml 1! tinyxml2 symbols are not distributed via apt for some reason, but you can manually download them and install via `dpkg -i` from here for bionic: https://launchpad.net/ubuntu/bionic/amd64/libtinyxml2-6-dbgsym/ .