moveit/moveit #2299: master vs. noetic-devel branches

rhaschke, 2020-09-07:
With today's release, the `master` and `noetic-devel` branches got synced (and should stay synced until we decide for an API freeze by the end of the year). This poses a new issue with our docker images: While the `master` branch was built against ROS Melodic so far, `noetic-devel`, obviously, should be built against Noetic. Now, that both branches got synced, master's docker image is building against Noetic as well, [causing some downstream projects to fail on Travis](https://travis-ci.com/github/ubi-agni/moveit_task_constructor/builds/183127482).

I think we should build the `master/noetic-devel` branch against both, Melodic and Noetic and provide corresponding, separate images. However, currently, our docker images are bound to a specific source branch (because [`.docker`](https://github.com/ros-planning/moveit/tree/3cc363372a6e52d378e6907096979ccff49cdccf/.docker) is part of the source tree). Hence, it's not possible to have two sets of docker images corresponding to the same branch. I see two options to overcome this issue:

1. Branch off `master` and provide Melodic images there. To stay up-to-date, we should regularly rebase master onto noetic-devel,  which should become our new default branch. This will also require to rebase open PRs against noetic-devel (instead of master).
For now, I paused auto-builds on `dockerhub` for master and manually pushed the (old) master images.
2. (Better) Migrate the `.docker` folder into its own github repository and provide corresponding Dockerfiles for each MoveIt branch there. To ensure up-to-date images, we should trigger builds on `dockerhub` with a regular schedule.

v4hn, 2020-09-07:
> until we decide for an API freeze by the end of the year

Just to keep expectations straight: We agreed to *discuss* an API freeze end of the year, not "decide for" straight away.

Concerning the actual issue: I am not yet fully convinced we have a problem here. Right now, we are in a transition period to have downstream projects support building on Ubuntu 20.04. / noetic.
I believe the build you cited fails because the release is not yet synced, right?

I would prefer not to add complexity to the docker definitions.

rhaschke, 2020-09-07:
Sure, downstream projects need to be adapted to support 20.04 / Noetic (as well).
The problem is that we cannot test downstream projects against the `master`=`noetic` branch of MoveIt on both, Ubuntu 18.04 and 20.04, because, with the current config, we only have 20.04 images available.

rhaschke, 2020-09-11:
@davetcoleman: I'm waiting for your feedback on this issue.

rhaschke, 2020-10-14:
I have created the branch [`docker-master`](https://github.com/ros-planning/moveit/tree/docker-master) to provide the master docker images, based on Melodic. This branch should be rebased onto master from time to time.