moveit/moveit #2301: PlanningSceneInterface in master seems to ignore header frame and pose

pbeeson, 2020-09-08:
### Description

In the older moveit_msgs::CollisionObject message (i.e. melodic and kinetic), there was a single pose.  In the current MoveIt master branches, there hasbeen added mesh_poses[] and similar for added individual poses to frames.  In adding this, it seems that when filling out the planning scene message, PlanningSceneInterface (at least addCollisionObject()) now copies [0,0,0,0,0,0,0] into the top level pose, regardless of what value you enter for the pose in the CollisionMessage sent to addCollisionObject().  Similarly, regardless of the frame_id set in the header, it overwrites this with "world".  

### Your environment
* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.04
* Source or Binary build?
source
* If source, which branch?
master

### Steps to reproduce
Call addCollisionObject in a PlanningSceneInterface, with a CollisionObject that has a frame and a top-level pose.  Then run /get_planning_scene, and you'll ALWAYS get this:
```
 world: 
    collision_objects: 
      - 
        header: 
          seq: 0
          stamp: 
            secs: 0
            nsecs:         0
          frame_id: "world"
        pose: 
          position: 
            x: 0.0
            y: 0.0
            z: 0.0
          orientation: 
            x: 0.0
            y: 0.0
            z: 0.0
            w: 0.0
```

### Expected behaviour
The frame should be used, and the pose should be a pose in the frame.  According to the newest move_msgs comments: 
```
# A header, used for interpreting the poses
Header header

# The object's pose relative to the header frame.
# The shapes and subframe poses are defined relative to this pose.
geometry_msgs/Pose pose

......

# The collision geometries associated with the object.
# Their poses are with respect to the object's pose

# Solid geometric primitives
shape_msgs/SolidPrimitive[] primitives
geometry_msgs/Pose[] primitive_poses

# Meshes
shape_msgs/Mesh[] meshes
geometry_msgs/Pose[] mesh_poses
```

### Actual behaviour
The top level pose/header is basically ignored, and the values set in mesh_poses and primitive_poses are assumed to be in "world" frame, even if they are not.

pbeeson, 2020-09-08:
The description was edited as I recalled the changes backwards.  The header/pose were added recently to the CollisionObject message (mesh_poses[] were always there), but seem to have not been fully integrated.  Unsure if moveit supports the header and Pose, but PlanningSceneInterface certainly does not know about these values in its copying of message data.

felixvd, 2020-09-09:
This is because #2037 is not merged yet. If you would like to use the feature, you can speed it up by adding your review :)

pbeeson, 2020-09-09:
Oh great.  Thanks for the pointer.

pbeeson, 2020-09-09:
Actually, while #2037 is related to my issue, it's not actually the "because [it] is not merged yet".  Adding pose to the CollisionObject message in moveit_msgs repo, and stating that this pose is the base pose used, when that's not true is really bad practice.   I don't understand why moveit_msgs is adding in message fields and comments in those messages that state things that are simply not the case.

```
# The object's pose relative to the header frame.
# The shapes and subframe poses are defined relative to this pose.
geometry_msgs/Pose pose
```
Th above is in moveit_msgs master, but it simply false.  This is NOT used in MoveIt master.

pbeeson, 2020-09-09:
This is not a bug in moveit, but more of an issue with how moveit_msgs is allowed to merge unsupport message changes into master well before moveit functionality for those changes is merged.  Moving gripe to https://github.com/ros-planning/moveit_msgs/pull/69

closed 2020-09-09 by pbeeson