moveit/moveit #2302: MoveIt Setup Assistant fails while generating .srdf

VictoriaBentell, 2020-09-09:
Environment: Noetic; Ubuntu 20.04; binary install from ros-testing (ros-noetic-moveit) on Sep 8 2020

Using MoveIt Setup Assistant with the [UR5 robot](https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/ur_description/urdf/ur5_robot.urdf.xacro), I click "Generate Package" and it gets to 8% and then there's a popup saying
`Failed to generate folder or file: 'config/ur5.srdf' at location:
/home/minty/catkin_ws/src/universal_robot/moveit_config/config/ur5.srdf` and the generator stops. But when I open ur5.srdf it appears that it generated fine: 
```
<?xml version="1.0" encoding="UTF-8"?>
<!--This does not replace URDF, and is not an extension of URDF.
    This is a format for representing semantic information about the robot structure.
    A URDF file must exist for this robot as well, where the joints and the links that are referenced are defined
-->
<robot name="ur5">
    <!--GROUPS: Representation of a set of joints and links. This can be useful for specifying DOF to plan for, defining arms, end effectors, etc-->
    <!--LINKS: When a link is specified, the parent joint of that link (if it exists) is automatically included-->
    <!--JOINTS: When a joint is specified, the child link of that joint (which will always exist) is automatically included-->
    <!--CHAINS: When a chain is specified, all the links along the chain (including endpoints) are included in the group. Additionally, all the joints that are parents to included links are also included. This means that joints along the chain and the parent joint of the base link are included in the group-->
    <!--SUBGROUPS: Groups can also be formed by referencing to already defined group names-->
    <group name="arm_group">
        <chain base_link="base_link" tip_link="wrist_3_link"/>
    </group>
    <!--DISABLE COLLISIONS: By default it is assumed that any link of the robot could potentially come into collision with any other link in the robot. This tag disables collision checking between a specified pair of links. -->
    <disable_collisions link1="base_link" link2="shoulder_link" reason="Adjacent"/>
    <disable_collisions link1="ee_link" link2="wrist_1_link" reason="Never"/>
    <disable_collisions link1="ee_link" link2="wrist_2_link" reason="Never"/>
    <disable_collisions link1="ee_link" link2="wrist_3_link" reason="Adjacent"/>
    <disable_collisions link1="forearm_link" link2="upper_arm_link" reason="Adjacent"/>
    <disable_collisions link1="forearm_link" link2="wrist_1_link" reason="Adjacent"/>
    <disable_collisions link1="shoulder_link" link2="upper_arm_link" reason="Adjacent"/>
    <disable_collisions link1="wrist_1_link" link2="wrist_2_link" reason="Adjacent"/>
    <disable_collisions link1="wrist_1_link" link2="wrist_3_link" reason="Never"/>
    <disable_collisions link1="wrist_2_link" link2="wrist_3_link" reason="Adjacent"/>
</robot>
```

Here's my terminal output:
```
roslaunch moveit_setup_assistant setup_assistant.launch 
... logging to /home/minty/.ros/log/6af7f81e-f22f-11ea-9351-f1503dd999ed/roslaunch-minty-41758.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/minty/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://minty:33403/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.8

NODES
  /
    moveit_setup_assistant (moveit_setup_assistant/moveit_setup_assistant)

auto-starting new master
process[master]: started with pid [41766]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 6af7f81e-f22f-11ea-9351-f1503dd999ed
process[rosout-1]: started with pid [41776]
started core service [/rosout]
process[moveit_setup_assistant-2]: started with pid [41779]
[ INFO] [1599609632.532601194]: Loaded ur5 robot model.
[ INFO] [1599609632.533346034]: Setting Param Server with Robot Description
[ INFO] [1599609632.535187401]: Robot semantic model successfully loaded.
[ INFO] [1599609632.535211203]: Setting Param Server with Robot Semantic Description
[ INFO] [1599609632.544267713]: Loading robot model 'ur5'...
[ INFO] [1599609632.544319736]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1599609633.047242510]: Stereo is NOT SUPPORTED
[ INFO] [1599609633.047298476]: OpenGl version: 4.6 (GLSL 4.6).
[ INFO] [1599609633.147818785]: Loading robot model 'ur5'...
[ INFO] [1599609633.147842880]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1599609633.496741669]: Loading Setup Assistant Complete
[ INFO] [1599609637.267220803]: -------------------------------------------------------------------------------
[ INFO] [1599609637.267266237]: Statistics:
[ INFO] [1599609637.267278083]:     11 : Total Links
[ INFO] [1599609637.267321132]:     55 : Total possible collisions
[ INFO] [1599609637.267344642]:      0 : Always in collision
[ INFO] [1599609637.267355395]:      3 : Never in collision
[ INFO] [1599609637.267363076]:      0 : Default in collision
[ INFO] [1599609637.267370624]:      7 : Adjacent links disabled
[ INFO] [1599609637.267380696]:     45 : Sometimes in collision
[ INFO] [1599609637.267389471]:     10 : TOTAL DISABLED
[ INFO] [1599609637.267434291]: Thread complete 28
[ INFO] [1599609664.124965870]: Updating kinematic model
[ INFO] [1599609664.125411447]: Loading robot model 'ur5'...
[ INFO] [1599609664.125482736]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ERROR] [1599609682.432995207]: Failed to generate entire configuration package

```

I'd love to contribute to resolving this, but I don't know where to even begin.

VictoriaBentell, 2020-09-09:
My workaround is to generate it the first time, and after it fails, move ur5.srdf into a higher folder and delete the rest of the generated files. Then under `Files to be generated` uncheck `config/ur5.srdf` and hit "Generate Package". Then move the original ur5.srdf back into the config folder.

rhaschke, 2020-09-09:
Thanks a lot for reporting. That was yet another issue introduced by the transition of `srdfdom` to `tinyxml2`. Sorry about that.
Will be fixed with the 0.6.2 release: https://github.com/ros/rosdistro/pull/26508

closed 2020-11-24 by VictoriaBentell