moveit/moveit #2316: errors occur after putting MFC and kinova controling code together

207900368, 2020-09-15:
### Description
Before I put the MFC  application and kinova controlling application together,I have generated the kinova controlling application sucessfully,and it can control the robot sucessfully.In the VS MFC appliction,I cited the generated kinova controlling application. Then many errors occur.Most of them are error 3254 and error 2838
[image]

my environment
windows10 vs2017 cmake conan

### Expected behaviour
I don't know why these errors occur and how to debug

felixvd, 2020-09-15:
I fail to see the connection to MoveIt. Maybe you should take this question to a Kinova support platform or to ROS Answers, with more detail.

felixvd, 2020-09-24:
This doesn't look like a MoveIt problem. Please reopen and add more detail if you think it is.

closed 2020-09-24 by felixvd